arduino/arduino-ide #245: can't chose variant and upload-method

mrbbp, 2021-03-19:
**Describe the bug**
in beta4,with stm32 distrib from STM32duino, i cant chose my variant or upload method, all the sub menus are not here

**To Reproduce**
1. install dan Drown stm32 version (not official -> https://dan.drown.org/stm32duino/package_STM32duino_index.json)
2. Go to 'Tools"
3. chose an STM Board
3 there no item in the menu to chose variant or upload method, etc etc...

**Expected behavior**
A complete menu. it was working on beta3

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Macos 
 - Version: 10.14.6

mrbbp, 2021-03-19:
how it was looking on Beta3

[image]

ubidefeo, 2021-03-19:
hi @mrbbp 

thank you for reporting this.
It looks like as we redefined out we gather platform information we introduced a regression.
Could also be part of the cleaned up gRPC protocol

@silvanocerza @kittaakos this one is for the two of you to figure out :)

mrbbp, 2021-03-20:
https://github.com/arduino/arduino-ide/issues/244#issuecomment-803430647

when i use b3 to chose my board , in b4 the submenu come back, but if change the board in b4, menu disappear
#244

ubidefeo, 2021-03-20:
@mrbbp 
in b4 we have changed a few things in how the IDE gets platform's information from the CLI, 
and also the gRPC interface has gone through some changes.

I think this has introduced some edge cases like this one, but we have enough information to know where to dig and fix it soon

The thing you encounter when switching from 4 to 3 and back to 4 is because the 3 has a different CLI version and I think it restores some information, but can't be sure because I do not write that code :)

ubidefeo, 2021-03-23:
@mrbbp 
the fix has been merged, you'll find it in tomorrow's nightly build.
Thank you for helping us solve a bug 🙏🏼

kittaakos, 2021-03-23:
> you'll find it in tomorrow's nightly build.

It won't work. That build will still contain the `0.17.0` CLI. The IDE does not pick up the most recent nightly.

https://github.com/arduino/arduino-ide/blob/f6e623ca9c703f255138e8d520ec06a6d3d9b103/arduino-ide-extension/package.json#L124

ubidefeo, 2021-03-23:
@kittaakos 
you are right...
I'm burned today, can't think straight.

@mrbbp the fix has been applied to the CLI, but the Nightly Build has of course a pinned release of the CLI, so it won't drag in the CLI Nightly.

Thank you, Akos

kittaakos, 2021-03-23:
I will take care of the CLI version update soon and notify this issue once the fix is available.

mrbbp, 2021-03-24:
Thank you for your work...

é.

Du bout du doigt...