arduino/arduino-ide #247: When trying to select text from the serial monitor it start selecting the text from the beginning and not the section you want

Macka323, 2021-03-20:
**Describe the bug**
When trying to select text from the serial monitor it start selecting the text from the beginning and not the section you want

**To Reproduce**
1. Program the Arduino to output serial data like letters
2. Open the serial monitor

**Expected behavior**
To select the text you only want not from the beginning to where you selected

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version: [e.g. 2.0.0] beta 4

per1234, 2022-02-06:
For the benefit of future readers, I'll add a note that the screencast shared here makes it a little easier to understand what this bug is;
https://github.com/arduino/arduino-ide/issues/167#issue-821042755

per1234, 2022-03-13:
I am not able to reproduce the exact behavior from https://github.com/arduino/arduino-ide/issues/167#issue-821042755 now, but I do see something similar.

When the mouse cursor crosses the left border of the Serial Monitor output field, all the text from there up is selected. This makes it difficult to get the leftmost character on line if you are selecting in an upwards direction.

When the mouse cursor crosses the right border of the Activity Bar, all the text in the entire Serial Monitor view, including the input field text (e.g., "**Message (Ctrl + Enter to send message to '\<board name\>')**"), the line ending menu text, and the baud rate menu text, and also the line numbers in the gutter of the editor view are selected.

[image]

ohault, 2025-04-04:
Can you please fix this bug ?
It exists for several years!