arduino/arduino-ide #262: Complete list of examples are not always shown

akash73, 2021-03-24:
**Describe the bug**
At the first launch, the build in examples and the library examples menu is correctly shown.
after selecting a random sketch from the examples list, then the menu shows only the built in examples

**To Reproduce**
Steps to reproduce the behavior:
1. Go to File  -> Examples (all examples are displayed)
2. Click on any examples
3. Go to File  -> Examples (only built-in examples are displayed)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MACOS
 - Version: [10.15.7]
First launch
[image]

same manu after opening any sketch
[image]

OS details
[image]

kittaakos, 2021-03-24:
Did you select a board, @akash73? You have to select a board to see the examples.
Here is a related question: https://github.com/arduino/arduino-ide/issues/192#issuecomment-796614926
Let me know if you cannot see the examples after selecting a board.

akash73, 2021-03-29:
@kittaakos thanks .

after selecting the board, 

The first time I select File -> Examples ,  the full list is not displayed.
The 2nd time , selecting again File-> Examples to see the full list.

tigoe, 2021-03-30:
A note on this: it seems that the full list only appears if you have a board and port selected. 

To reproduce:
1. unplug all boards
2. Quit
3. restart
4. Check examples menu.

In this case, you'll see only the core examples, not the libraries examples. I expect to see the libraries examples as well. I get that they are filtered by board, but I think that filtering could happen after you pick the board, not before.

ubidefeo, 2021-03-30:
@tigoe 

>I get that they are filtered by board, but I think that filtering could happen after you pick the board, not before.

I can't remember why it is this way, but I'll check what can be done 👍🏼

tigoe, 2021-03-30:
My guess would be that the menus are context sensitive to the board type.
But in the old ide there's never a time when you don't have a board type,
because the Uno is the default. So the error never occurred.

per1234, 2021-03-31:
I have a lot of libraries installed and it takes about a minute between selecting a board and the library examples showing in the **File > Examples** menu.

tigoe, 2021-03-31:
@per1234 You might be outside the norm for a beginner. You might even be
outside the norm for an advanced user. It takes about two seconds for me,
and I thought I had a lot of libraries.

Many teachers count on the libraries showing up in the list as they did in
IDE 1.0, so I would suggest finding a way to make it work even when you
don't have a board selected.  Perhaps a list item saying "Pick a board to
see matching examples from your installed libraries"

SofiaEngvall, 2022-10-12:
I off course want to see all the examples all the time. If this is not the default behavior there should be an option for it. I am currently looking at examples to decide what more boards to buy. I don't want to go through all models of boards (200+ I believe) installed to be able to see all examples! As a workaround I installed an older version of Android studio (1.6.5) and there I can see everything. This should not be necessary, right? Thank you! :)

SofiaEngvall, 2022-10-12:
> @SofiaEngvall there seems to be a misunderstanding. The intended behavior of Arduino IDE 2.x is exactly the same as 1.6.5. The unresolved component of this issue is a **BUG** that causes library examples to occasionally not be shown under the **File > Examples** menu. The solution is to fix the bug, not to add an option. If you experience the bug, simply open a new window. That almost always fixes the problem for me.

Thank you for clarifying. The reason I reacted as I did was since you used the word **solution** instead of **workaround**. Great that you're aiming for a full solution.

I'm actually writing a new bug report now and I'm curious how this version could be the suggested download on the Arduino web page it there are these much basic functionality bugs? It really seems like it's not ready yet. I'm still typing there but I'm thinking maybe I could contribute?