arduino/arduino-ide #268: Conditional breakpoint

rei-vilo, 2021-03-25:
**Describe the bug**

Conditional breakpoint only stops the first time.

Adding `i%2` to line 8 only stops the first time when i=1, but not after when i=3 or 5.

**To Reproduce**

Same example as for #120 but with modified shown below.

**Expected behavior**

Debugger stops at 1, 3 and 5.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**

 - OS: Windows 10
 - Version: 2.0.0 beta 4
 - SAMD core 1.8.11

**Additional context**

+ Modified **LocalLibrary.cpp**

```
#include "LocalLibrary.h"

// Code
void blinkLED(uint8_t pin, uint8_t times, uint16_t ms, bool level)
{
    for (uint8_t i = 0; i < times * 2; i++)
    {
        digitalWrite(pin, (level ^(i%2)) ? HIGH : LOW);
        delay(ms >> 1);
    }
}
```

ubidefeo, 2021-04-06:
@rei-vilo 
I have just tested this and it is working for me
I use an `uint8_t` and only break on `counter % 2 == 0`

per1234, 2022-12-02:
Closing as fixed by https://github.com/arduino/arduino-ide/pull/1706

After quite some deciphering of this unnecessarily complex report, I finally determined that it is the same as https://github.com/arduino/arduino-ide/issues/412

closed 2022-12-02 by per1234