arduino/arduino-ide #269: Release 2.0.0 beta 4 very slow

rei-vilo, 2021-03-25:
The release 2.0.0 beta 4 is very slow compared to previous releases and other similar IDEs.

- At launch
- During normal operation

stevisco, 2021-03-26:
hi @rei-vilo  could you tell us something about your operating system and configuration ? are you comparing to beta3 and you saw beta3 was faster ? thanks!

rei-vilo, 2021-03-26:
Sure, I'm using Debian with Gnome on a i5-8259U with 8GB of RAM.

Beta 4 is slower than beta 3, and Beta 4 is much slower than Visual Studio Code running on the very same configuration, albeit both Eclipse Theia and Visual Studio Code are built on the same underlying technologies.

ubidefeo, 2021-03-28:
@rei-vilo 
would you be able to specify what is slow?
there aren't significant changes between Beta 3 and 4 which should make it sluggish.
Are you experiencing UI slowness?
Which parts seem to be slower in 4 than 3?

thank you
u.

rei-vilo, 2021-03-30:
Some unscientific measures:

+ Arduino IDE 2.0.0 beta 4: 34 s on first launch then 13 s on subsequent launches
+ Visual Studio Code 1.54.3 with Arduino extension 0.4.0: 8 s on first launch then 3 s on subsequent launches

ubidefeo, 2021-03-30:
@rei-vilo 
They are 2 different products, also the Arduino extension for VS code does a lot of things, but it's not based on our CLI.
The IDE is a front-end to a CLI backend, and it might take some time to get all the pieces working.
You cannot expect them to work the same.

The difference in speed between 3 and 4 is something I'm curious about, though

rei-vilo, 2021-05-24:
Seems to be also reported at #385

closed 2021-05-24 by rei-vilo