arduino/arduino-ide #270: Retry upload

CreativeRobotics, 2021-03-25:
I get frustrated by the frequency with which USB bootloaders fail to work, or when my own sketch caused the USB to crash, and I have to either reset, or force a board into the bootloader. What makes it worse is having to wait whilst the IDE recompiles before uploading again.

I would like a 'retry upload' button or menu item and shortcut that will attempt to re-upload the last compiled sketch without going through the whole compile process.

ubidefeo, 2021-03-26:
@CreativeRobotics 
in theory anything already compiled is cached and the time when doing `upload` again should be much faster than the first one.
Which platform are you targeting?

CreativeRobotics, 2021-03-26:
I'm using mostly the Adafruit and Mattairtech SAMD board definitions and the SAMD21 MCU.
I know it is faster when you retry an upload after it has failed, but it still takes a long time to get to the point where the IDE claims it is trying to upload.

The fact that the IDE always reports that it is compiling before uploading, even after a previous failed attempt at uploading, made me assume that some recompilation is still going on. If thats not happening, and it is just re-trying the upload, then the IDE shoudn't report that it is compiling first.

If it is performing an recompilation, even with caching, then a retry upload option would just skip that step, and in my case save me sitting and waiting for 20+ seconds.

closed 2021-03-26 by per1234