arduino/arduino-ide #275: Cmd+W should close tab, not window

joshka, 2021-03-26:
**Describe the bug**
Arduino acts as a container for tabs. Most people see this most often in browsers or other editors. The keyboard for closing a tab in those cases is Cmd+W (Ctrl+W on Windows). In Arduino IDE this closes the entire window.

**To Reproduce**
Steps to reproduce the behavior:
1. Press Cmd+W

**Expected behavior**
Close the current tab

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macos
 - Version: 2.0.0-beta-4

**Additional context**
This could be a feature request or a bug, or a highly opinionated preference. I'm not sure which.

ubidefeo, 2021-03-26:
hi @joshka 

>This could be a feature request or a bug, or a highly opinionated preference. I'm not sure which.

I believe it's an opinionated preference :D
The Arduino IDE's behaviour has traditionally been to keep all files of a Sketch open (the Sketch is a folder containing a `.ino` file with the same name of the folder and a bunch of other files to complete the project).
As you noticed, you cannot close tabs belonging to a sketch.

If you end up with an open tab as result of going to a definition or while using the Debugger you will be able to close such tabs because they don't believe to the Sketch.
Such tabs will present a closing `X` and will be closed by a CMD/CTRL+W.

We have long debated wether IDE 2.0 should behave as the Java IDE and the current behaviour (as annoying to myself it can be) has been decided to be brought into the new IDE.
I'll mark this one as an enhancement, and if it gathers enough 👍🏼 we'll bring it back to the discussion table.

Thank you
u.

closed 2021-03-26 by per1234

joshka, 2021-03-27:
Makes sense, thanks for the thoughtful response on this. Definitely an opinionated preference :)