arduino/arduino-ide #279: Quicker upload notification

tigoe, 2021-03-27:
Couldn't find an issue on this one, so here goes:

1. Open new sketch
2. Choose board and port
3. Click upload button
4. 
After "Compilation complete", the wait for notification about upload takes many seconds, long enough for the user to think something has gone wrong. IN that case, they might hit the reset or unplug, which disrupts the upload. 

If you turn on verbose notices for upload you get

````
Compilation complete.
Performing 1200-bps touch reset on serial port /dev/cu.usbmodem1431301
Waiting for upload port...
````

Which is helpful, but the reams of text you get otherwise can be bothersome. I think it would be good to offer short "Connecing to board..." then "Uploading..." progress messages so the user knows what the status is and doesn't disrupt it.

ednieuw, 2021-04-23:
Same annoyance. I can not see that the program is compiling or uploading.

tigoe, 2021-09-05:
Update using 2.0.0-beta.11

This is still unchanged. The delay is long enough that I don't feel that I can use 2.0 to teach beginners yet.

per1234, 2022-08-02:
Closing as resolved by https://github.com/arduino/arduino-ide/pull/1260

closed 2022-08-02 by per1234