arduino/arduino-ide #283: Tab list opens with an offset or doesn't display at all

Lorandil, 2021-03-28:
**Describe the bug**
When clicking on the tab list drop down in the top right of the IDE window (below serial monitor), the list of files appears to the right of the IDE window with some large offset making it hard to find, or sometimes not at all

I'm running Windows 10 Home and a dual monitor configuration (horizontal desktop extension, each monitor is FHD).

**To Reproduce**
[image]

1. start beta4
2. un-maximize the window
3. click on the tab list arrow
4. the window appears outside the main window too far to the right

or in dual monitor setup:
1. start beta4 on monitor 1
2. maximize the window
3. click on the tab list arrow
4. the window appears some 100 pixels inside the second monitor (or sometimes not at all if there is e.g. a browser window)

**Expected behavior**
The window should appear inside or at least touching the main window. Ideally it would show below the drop down arrow.

**Screenshots**
yup!

**Desktop (please complete the following information):**
 - OS: Window 10 20H2 home 64 bit
 - Version: 2.0 beta4
 - dual screen setup (not sure if it matters)

per1234, 2021-03-28:
We discussed a context menu offset issue in https://github.com/arduino/arduino-ide/pull/258#pullrequestreview-618643537. It sounds like there is a possibility that PR will end up fixing the issue.

Lorandil, 2021-03-29:
I read #258.
I think that I did change the editor font size with CTRL + mousewheel in beta3,
but I was unable to change the font size in this way in beta4. Not sure if this helps.