arduino/arduino-ide #284: Option to add existing files to a sketch

Lorandil, 2021-03-28:
If I want to reuse some code I wrote earlier without creating a library, I copy those files to the sketch folder.
At the moment I have to restart the IDE to integrate them into the sketch.
It would be great if the files drop down menu would allow to "add existing files"

ubidefeo, 2021-03-28:
@per1234 
This one is something I have also found myself in need of.
What's your opinion?

per1234, 2021-03-29:
I don't think it would do any harm. This tabs menu is not something a beginner needs to use, so adding additional functionality to it doesn't add complexity to the beginner's UI.

ubidefeo, 2021-03-30:
thanks @per1234 

let's create a task for this :)

per1234, 2022-12-24:
Thanks to [a forum user](https://forum.arduino.cc/t/adding-files-change-of-behavior-bug-or-me-being-dumb/1068993), I just discovered that Arduino IDE 1.x has this capability, so this is actually a matter of feature parity.

I remember that when the **Sketch > Add File...** capability was added to Arduino IDE 2.x, it was specified as a way to add supplemental files (e.g., documentation) to the `data` subfolder of the sketch. I never used this feature in Arduino IDE 1.x, but now that I try it out, I see the Arduino IDE 1.x behavior is more complex than what was specified for the Arduino IDE 2.x implementation:

- If the selected file has a supported sketch code file extension (e.g., `.ino`, `.h`, `.cpp`, `.c`, `.S`), add it to the the root of the sketch folder.
- If the selected file has any other file extension, add it to the `data` subfolder of the sketch.

Arduino IDE 2.x's **Sketch > Add File...** capability always does the second of these, even when adding a file with a supported sketch code file extension.

PR #1805 (2023-01-11): fix: aligned `Add File...` behavior with IDE 1.x

### Motivation

As asked for in #284, IDE2 can now add source files to the sketch folder root. Other files go into the `data` folder in the sketch folder root.

### Change description

 - code files will be copied to the sketch folder root
 - other files go under the `data` folder in the sketch folder root
 - after adding code files to the sketch, new files are expected to open in an editor.

https://user-images.githubusercontent.com/1405703/211023968-00185533-0f4f-426a-8e7c-104dca267070.mp4

### Other information

Closes #284

### Reviewer che [... cut]
commit 197cea2a (2023-01-11): fix: aligned `Add File...` behavior with IDE 1.x

 - code files will be copied to sketch folder root
 - other files go under the `data` folder in the sketch folder root

Closes #284
closed 2023-01-11 by kittaakos