arduino/arduino-ide #287: Serial power re-enumerates after upload with SAMD boards (probably other USB native)

tigoe, 2021-03-29:
When you upload a new sketch to a SAMD board, the serial port list drops your port choice and chooses the one above it when the board is reset for upload.

Steps:
1. Open new sketch
2. Choose board and port
3. Upload to a SAMD board
4. Note the serial port choice after upload. It will be the one above your port in the list. 

This causes confusion when you try to open the serial port. You can open the port, but you have to re-choose it first. 
MacOS Big Sur 11.2.3 20D91

This is related to an earlier issue...

tjpetz, 2021-03-29:
I have this same problem with the mbed based Nano 33 BLE Sense. I’m also using MacOS Big Sur. 

It’s a particular challenge when using 2 boards simultaneously as after upload the active window will switch to the other port. This makes it particularly difficult to see any serial data right after the load. 

Also resetting the Arduino board will cause the same rescan and thus confusion on which port is to be used.

Xtreemtec, 2021-04-05:
Running a SAMD21 Zero all day on windows and while programming it enumourates a different port. ( but we are used to this as programming port always has a different comm port)  but nicely jumps back for serial monitor.    So i cant relate to this for windows 10 at this point.

ubidefeo, 2021-04-08:
@tigoe 
what's the process to create such virtual serial port?
I'd like to replicate the setup to test

closed 2022-08-24 by davegarthsimpson