arduino/arduino-ide #289: Tear off serial monitor window (and maybe others)

Lorandil, 2021-03-29:
In IDE 2.0 the serial monitor window is located at the bottom of the main window.
I like to keep my code window clean, so with IDE 1.8.13 I move the serial monitor to the second screen.
This allows me to watch outputs while making changes to the code without wasting screen space.
The functionality is missing in IDE 2.0 beta4.

It would probably make sense for other windows (output), too.

ag88, 2021-03-30:
a tear off serial monitor is rather useful, but i'd think a 'better' option is to delegate it to a dedicated serial monitor app like putty
https://www.putty.org/
good serial monitors with a lot of features take a lot of effort to develop. it would be better to find existing solutions that already works well
a problem though is for a 'serial monitor' that integrates with a graphing tool. that i'd guess could be met with a dedicated tool as well.

Lorandil, 2021-04-01:
Thanks for the tip.
The problem with external tools like putty is, that they block the serial port.
The IDE won't be able to upload to the micro controller until the connection is manually closed.
With the integraded serial monitor the IDE can claim and release the port as necessary.

Xtreemtec, 2021-04-05:
Started today with 2.0 and this was the first thing i missed..   We run a triple screen setup for programming and our debug is always on a second screen as we normally have a lot of data coming from our processors while programming. 
Also that CTRL + ENTER to send is very annoying..   But i will look if there is an issue open for that too.

ironhal0, 2021-05-28:
I started with 2.0 today, and this was also the first thing that I missed. Any apps that don't let me tear off individual functions to separate screens are essentially hobbling my 3 monitor setup and I take a big step backward in efficiency.  Being able to watch the serial output in its own large window while keeping code open in a separate large window is essential.

DaveTur, 2021-09-13:
It would also be very useful to 'Tear off' or 'float' the Debugger area.  Being able to tear off the individual debugging windows such as CALL STACK and WATCH would be even better.

PerryBebbington, 2021-12-01:
+1 for this, for all the reasons others have already given.

tomjennings, 2022-01-05:
I'm a heavy user of serial monitor. 1.8.19, vs 2.0.0 here. My projects are sometimes large with voluminous runtime output during development phase. Example below. "Tear off" or somehow separate window is critically important, especially when I'm field-debugging with a laptop and small screen. 

Here's a snapshot of my current project's output. Visualization of time series data is easy with dumb old prints. (The 2.0 serial monitor also suppresses multiple spaces but that is a separate Issue.) 

[image]

CreativeRobotics, 2022-03-15:
I agree, having the Serial monitor at the bottom is a massive step back in useability. I'm testing out the IDE RC5 and I'm still not seeing any compelling reason to use it over the old IDE, its slower, cluttered and clunky.

fstasi, 2022-03-17:
@ironhal0 it's definitely in our roadmap and at the top of the list.

We are giving maximum priorities to bugs that compromise the standard usage, next will come performance and UX improvements, like this issue with the monitor. We already designed the new one that is gonna be a separate window like the serial plotter.

Digitalwand123, 2022-04-04:
Regarding the Serial Monitor pop-up in a separate window -- I read Arduino comments on the benefits of not doing so, but there are also many reasons of indeed creating Serial monitor in a separate window.   Often I have situations where I do need full windows to review data etc.

-- Why not have a flag to enable one option or the other.

On a side note, I noticed that also Manage Libraries is squeezed into main windows.  I think the 1.8 version is more useful, but I can "live" with it.  I believe the Serial Monitor pop-up is important -- at least to me

PaulVdBo, 2022-07-26:
A detachable serial window is on top of my wish list for ide 2.0 !
For the rest RC9 looks good !
thanks !

kittaakos, 2022-09-06:
https://github.com/eclipse-theia/theia/commit/a07b74644372ebc6c5f04e153bd7d8607a5afc5c

RuudMR, 2022-09-15:
PLEASE:
I also want to:
Bring back the separated serial monitor from the IDE 1.xx version to the Version 2.0.
So, i keep using the 1.xx version and not the 2.xx version. (als with all the students i have.)
Please bring it back!! 

Ruud, Arduino teacher.

marmilicious, 2022-09-24:
Like the new editor. Totally miss being able to have the serial monitor as a separate window on a second monitor though. Looking forward to that update. Thank you devs.

CreativeRobotics, 2022-10-05:
Why is this not fixed yet!

Its a serious design flaw in the software and makes it less easy to use than the previous version. I just started using the 2.0.0 release and apart from improvements in speed of compilation, the serial port and the way boards and libraries are managed in such a messy and cluttered way feels like a massive retrograde step.

At the very least, the IDE should respect the users desire to watch the serial port during an upload, but it forces you to switch to the Output tab every time it prints a line.

It also shouldn't conflate COMM ports with boards. They are not always the same thing.

I'm constantly using the old IDE on a large monitor with it side by side with the serial port. his new IDE is just harder to use.

ironhal0, 2022-10-05:
Hate to be a naysayer, but if the current dev team can’t in the span of TWO YEARS address such a fundamental design flaw that affects all
users, it leaves very little hope for the future of this IDE. I’d suggest checking out one of the many alternatives, like PlatformIO or Eclipse.

star297, 2022-10-05:
I've moved over to PlatformIO.
I don't think I would go back even if the Arduino dev guys do finish the job.
Sits on to top of VS Code, takes a while to set up, performance is good and loads of options.
Don't think it has a tear off serial monitor but you can move it to the side which suits me anyway.
You can have more than one monitor, I have one for code build output and another to monitor the device serial port.
Github source control is handy to see code changes as you go, not done much with that part yet.
I wouldn't say it was a professional IDE, but not far off and easier to set up.

cshamis, 2022-10-28:
I would love to see the **serial port exposed over a TCP port**, so I can just telnet to it.  Then I could use any other thing I wanted to scroll search browse, capture... heck I could even use a separate laptop just to scroll the logs...

jpmeijers, 2023-01-14:
It is really frustrating that the serial monitor is hidden and the output tab is shown when I upload data to a device. When I click on the serial tab during build/upload, the IDE switches back to the output tab. This is really frustrating, and also prevents me from clearing the serial monitor before the device is done uploading.

If the serial monitor can be in a separate window like in IDE 1.8, or in a split editor tab, one should be able to clear the contents and view data immediate after a device reset, without having to keep on trying to switch to a tab and the IDE switching back.

> We are giving maximum priorities to bugs that compromise the standard usage

@fstasi This is a critical issue as it prevents me from properly developing with serial output.

wdoekes, 2023-01-21:
Yes, I concur. It feels like the developers aren't using the IDE themselves. Or at least not using the serial monitor like the rest of us.

If someone could get it to stop switching/focusing tabs on output, that would be a great relief.

ubidefeo, 2023-01-21:
@wdoekes 

> It feels like the developers aren't using the IDE themselves

Do you realise how absurd that sounds?
We use the IDE, and so do many inside Arduino.
All of us sometimes experience some of our users' frustrations.

Give us the time to bring in the goods.
React with a 👍🏼 to an issue, we watch those

Cheers
ubi

ironhal0, 2023-01-22:
> > It feels like the developers aren't using the IDE themselves
> 
> 
> 
> Do you realise how absurd that sounds?

Not quite as absurd as it sounds for the v2.0 IDE to still have a non-usable serial monitor for three years after its initial alpha release.

The dev team is obviously using the IDE in some manner, but if you were living in the serial monitor in the way you’re required to live in the serial monitor while doing any real embedded development, there is no way you would suffer through three years of a gimped serial monitor. This would have been prioritized long before now if you were.

Xtreemtec, 2023-01-22:
It really bugs me that this is still not fixed.   I think it was 2,5 years ago that I posted a reaction in this tread and still it is limited.  I think it is due to change to electron or what is it based on now that they are not able to easy open a separate window.     Well at least let me make the arduino software then twice as wide ( 2 screens ) and let me put the serial terminal to 1 screen.  We use the serial terminal a lot and with much data going trough that we always have it open full screen.  And an external serial monitor program doesn’t work as you have to remember to open and close the port all the time.  This is the main reason we still not using ide 2.  It is installed and we sometimes update to see if it changed.   But still running 1.8 as it’s serial port detaches.   While unreality would like to go to 2 due to we running code on the Zero.  So debug with the ICE would be a nice option.  Daniel

CptHolzschnauz, 2023-01-22:
Peace! Fingerpointing is not engineering. Something like https://ioninja.com/plugins/serial-monitor.html would be cool. 
My wishes: A search field and a "triggerword" field e.g. when the triggerword appears in the monitor, it stops the autoscroll and a sound plays. 
Maybe a seperate open-source based programm that can be run paralell to IDE 1 and 2 as well would be a solution?
Or the IDE 1 serial monitor pimped up as a fork? Any hints to a existing project, ideas etc. ?

CreativeRobotics, 2023-01-22:
> > > It feels like the developers aren't using the IDE themselves
> > 
> > 
> > Do you realise how absurd that sounds?
> 
> Not quite as absurd as the v2.0 IDE still having a non-usable serial monitor for three years after its alpha release.
> 
> Yes obviously the dev team is using the IDE in some manner, but if they were living in the serial monitor in the way you’re required to live in the serial monitor while doing any actual embedded development, there is no way they would suffer through three years of a gimped serial monitor. This would have been prioritized long before now if they were. v1.8 is still infinitely more usable strictly because of this alone.

I agree, every time I use the new IDE I feel like I'm beta testing. I just spent an hour trying to figure out why I was getting zero serial data from an ESP32 board only to discover that it had just decided to stop working. I only established it was the app when eventually I decided to quit it, and discovered that the close button was not responding. I had to force quit and restart the app, which then took over a minute to launch again.

Sorry to be blunt but for me the only good side to this new IDE is the CLI, and the possibility that someone else will make a proper IDE to sit on top of it.