arduino/arduino-ide #295: Incorrect context menu position when the editors are in split mode.

kittaakos, 2021-04-01:
Originally from [here](https://github.com/arduino/arduino-ide/pull/258#issuecomment-809460412):

> Also, please add the following to fix the issue when there are multiple "sketch-control" toolbar items. Locating them via their ID is flawed.

Bug:
https://user-images.githubusercontent.com/1405703/113262644-21557180-92d1-11eb-9103-0ab0a16f2c78.mp4

Fix:
```diff
diff --git a/arduino-ide-extension/src/browser/contributions/sketch-control.ts b/arduino-ide-extension/src/browser/contributions/sketch-control.ts
index 6e4010c..db0aa48 100644
--- a/arduino-ide-extension/src/browser/contributions/sketch-control.ts
+++ b/arduino-ide-extension/src/browser/contributions/sketch-control.ts
@@ -6,6 +6,7 @@ import { ContextMenuRenderer } from '@theia/core/lib/browser/context-menu-render
 import { Disposable, DisposableCollection } from '@theia/core/lib/common/disposable';
 import { URI, SketchContribution, Command, CommandRegistry, MenuModelRegistry, KeybindingRegistry, TabBarToolbarRegistry, open } from './contribution';
 import { ArduinoMenus } from '../menu/arduino-menus';
+import { EditorWidget } from '@theia/editor/lib/browser';
 
 @injectable()
 export class SketchControl extends SketchContribution {
@@ -24,15 +25,28 @@ export class SketchControl extends SketchContribution {
     registerCommands(registry: CommandRegistry): void {
         registry.registerCommand(SketchControl.Commands.OPEN_SKETCH_CONTROL__TOOLBAR, {
             isVisible: widget => this.shell.getWidgets('main').indexOf(widget) !== -1,
-            execute: async () => {
+            execute: async widget => {
                 this.toDisposeBeforeCreateNewContextMenu.dispose();
                 const sketch = await this.sketchServiceClient.currentSketch();
                 if (!sketch) {
                     return;
                 }
+                let target: HTMLElement | undefined = undefined;
+                if (widget instanceof EditorWidget) {
+                    const tabBar = this.shell.getTabBarFor(widget);
+                    if (tabBar) {
+                        const elements = document.querySelectorAll(`#${SketchControl.Commands.OPEN_SKETCH_CONTROL__TOOLBAR.id}`);
+                        for (const element of Array.from(elements)) {
+                            if (!target) {
+                                if (tabBar.node.contains(element)) {
+                                    target = element as HTMLElement;
+                                }
+                            }
+                        }
+                    }
+                }
 
-                const target = document.getElementById(SketchControl.Commands.OPEN_SKETCH_CONTROL__TOOLBAR.id);
-                if (!(target instanceof HTMLElement)) {
+                if (!target) {
                     return;
                 }
                 const { parentElement } = target;
```

kittaakos, 2022-08-16:
It's impossible to split the editor or modify the layout in IDE2. I am closing as invalid.

closed 2022-08-16 by kittaakos

per1234, 2022-08-16:
I'll add some references to related items just to create linkages between them for better discoverability of all related information:

The split editor capability was intentionally disabled (along with ["Flexible Window Layout"](https://www.typefox.io/blog/flexible-window-layout-in-theia-ide)) by https://github.com/arduino/arduino-ide/pull/940 due to it not working correctly. The request to add a functional split editor capability is tracked at https://github.com/arduino/arduino-ide/issues/909