arduino/arduino-ide #296: Make editor detect external changes to open files in a sketch

Lorandil, 2021-04-01:
Sometimes a source fiIe that is open in the Arduino IDE becomes modified 
from an external program (e.g. an alternate text editor like notepad++, or a
bitmap in a header file which was converted to hex by tools like image2cpp).
Because the Arduino IDE doesn't detect these modifications, it is required to 
close and reopen the sketch or else changes might get lost.

I would like the IDE to detect the file change and ask what to do (like in notepad++).
This feature would help beginners, because it is frustrating if changes are ignored or
become lost.

There might even be a setting to automatically open the changed file (which will
become tricky, if there are unsaved changes).

ubidefeo, 2021-04-01:
>or a bitmap in a header file which was converted to hex by tools like image2cpp

that's how you sold it to me :)

ag88, 2021-04-02:
in eclipse to 'add missing files' or to get the workspace project updated, it is pressing F5 (refresh), i'm not sure how easy or difficult to implement similar

ubidefeo, 2021-04-06:
relates to #284

per1234, 2022-12-24:
As I explained in my previous comment, Arduino IDE already has this behavior so I will close this as resolved. If anyone finds that external changes to a saved sketch file are not reflected in the Arduino IDE editor when using the latest [nightly build of Arduino IDE](https://www.arduino.cc/en/software#nightly-builds), then please submit a new bug report and we will investigate it.

closed 2022-12-24 by per1234