arduino/arduino-ide #301: Stuck on splash screen with 2.0.4.beta.4 on macOS Big Sur 11.2.3

galexite, 2021-04-05:
**Describe the bug**
The latest IDE release does not work for me on macOS. The IDE will not pass the splash screen, and the splash screen will remain in the foreground until the application is quit.

**To Reproduce**
Steps to reproduce the behavior:
1. Drag Arduino IDE application off DMG and into the Applications folder.
2. Open the application in the Applications folder.

**Expected behavior**
The Arduino IDE starts and presents the workspace.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Big Sur 11.2.3
 - Version: 2.0.4-beta.4

**Additional context**
I have been working on my own application using the Theia framework on the same computer, and I have not had any issues. I have never used Arduino IDE 2.0 on this computer before. I have the traditional IDE installed and working.

galexite, 2021-04-05:
There are no (relevant) errors or warnings printed to the Developer Tools console, which I can open via ⌘⎋I.

per1234, 2021-04-05:
Hi @galexite. Thanks for your report.

Please try the latest nightly build of Arduino IDE 2.x:
https://www.arduino.cc/en/software#nightly-builds
A recent improvement to the IDE (https://github.com/arduino/arduino-ide/pull/266) adds error reporting in this situation. This isn't a fix. However, it provides a helpful error message that is very valuable for troubleshooting this type of problem, which can be caused by several diverse problems.

With the nightly build, you should see an error notification at the bottom right corner of the IDE when it is starting up. Please copy and paste the full text of that message here.

galexite, 2021-04-05:
> Hi @galexite. Thanks for your report.
>
> Please try the latest nightly build of Arduino IDE 2.x:
> https://www.arduino.cc/en/software#nightly-builds
> A recent improvement to the IDE (#266) adds error [...]

Hi @per1234, there is no change with this nightly build, I'm afraid. There is no error notification, and no change to the output in the console either.

[image]

galexite, 2021-04-05:
I tried starting the binary inside the application package from the Terminal, and this appears to be the issue:

```
daemon INFO Starting daemon from /Volumes/Arduino IDE 2.0.0-beta.4-nightly.20210405/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Using config file: /Users/georgewhite/.arduinoIDE/arduino-cli.yaml
daemon INFO arduino-cli version nightly-20210329
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon ERROR Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Failed to start the daemon.
daemon ERROR Error: Failed to listen on TCP port. Address already in use.
    at new DaemonError (/Volumes/Arduino IDE 2.0.0-beta.4-nightly.20210405/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:370:28)
    at Function.parse (/Volumes/Arduino IDE 2.0.0-beta.4-nightly.20210405/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:388:24)
    at Socket.<anonymous> (/Volumes/Arduino IDE 2.0.0-beta.4-nightly.20210405/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:281:57)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)
daemon INFO Restarting daemon in 5 seconds...
daemon INFO Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Daemon exited with exit code: 5.
daemon INFO Restarting daemon in 4 seconds...
daemon INFO Restarting daemon in 3 seconds...
daemon INFO Restarting daemon in 2 seconds...
daemon INFO Restarting daemon in 1 seconds...
daemon INFO Restarting daemon now...
```

I'll have a look and see where port 50051 is being used.

per1234, 2021-04-05:
Port 50051 is used by Arduino IDE for communication with the [Arduino CLI](https://arduino.github.io/arduino-cli/latest/) tool via a gRPC interface.

You can configure this port by opening `/Users/georgewhite/.arduinoIDE/arduino-cli.yaml` in a text editor and then editing the `daemon.port` configuration key you find there. After that, the IDE should be able to start.

Since there is another issue with discussion about improving on this experience of port conflicts: https://github.com/arduino/arduino-ide/issues/201 I'll go ahead and close this one as resolved. If you find that doesn't fix the issue, you're welcome to comment here and we'll investigate.

closed 2021-04-05 by per1234

galexite, 2021-04-05:
Port 50051 is in use by Canonical multipassd, which I use for Linux development.