arduino/arduino-ide #307: faulty warning

jerabaul29, 2021-04-09:
I am using the latest IDE2.0 on Ubuntu 20.04.

I am compiling a sketch for the Sparkfun Artemis board (third party boards). I installed the Sparkfun Artemis core v1.2.1 (installed by hand). The sketch compiles fine (see message in the command line interface pane).

However, I get some faulty error messages, in particular, see the one here:

[image]

Any idea:

- what is causing it?
- how to fix it?

mrbbp, 2021-04-12:
hello, i've got the same pb with added libraries

i use SD_FAT lib and have `unknown_type`, as it doesn't read the lib for var, knowned functions and authorized name

[image]
[image]
[image]

kgrabe, 2021-05-23:
to recreate on win 10, ide beta 7:
[image]
now comment out not-existent library, comile to get error:
[image]
now uncomment the libary..
[image]
and previousely ok code shows errors even though the code compiles ok:
[image]

per1234, 2022-06-23:
Closing as resolved by https://github.com/arduino/arduino-language-server/pull/92

closed 2022-06-23 by per1234