arduino/arduino-ide #308: Serial output window can not be resized smaller than 10 lines

ednieuw, 2021-04-11:
You can not  make the output windows smaller than 10 lines (with font size 14) as with the old IDE. 

**Expected behavior**
1) It should be able to minimize it to one line 
2) Make an option to removed the output window

**Desktop (please complete the following information):**
 - Windows 10 
 - Version:  2.0.0 beta 4

**Additional context**
 On laptops the editor screen has not many lines left.

kittaakos, 2021-04-12:
> 2\. Make an option to removed the output window

Have you tried to toggle it?

[image]

https://user-images.githubusercontent.com/1405703/114353086-e96dea00-9b6c-11eb-8172-d7b8ca13ab50.mp4

kittaakos, 2021-04-12:
> A short cut key for this toggle function and it is perfect

<kbd>Ctrl/⌘</kbd>+<kbd>J</kbd> should work. We can change the keybinding if required.

> 1. It should be able to minimize it to one line

I am going to reopen this task as I think this comment 👆 is useful. I hope the current workarounds are fine for the time being. Thanks for your help!

Lorandil, 2022-01-06:
Yes, please add a hide/toggle button :)
Or enable an auto hide feature like in Visual Studio...

closed 2022-10-07 by AlbyIanna