arduino/arduino-ide #310: Board Install - Access Denied

MikeSW17, 2021-04-12:
Fresh (first time) Install of IDE 2.0 beta 4

When attempting to install "Arduino MBed OS Nano Boards", it fails with "Access Denied"

` downloading arduino:arm-none-eabi-gcc@7-2017q4
downloading arduino:bossac@1.9.1-arduino2
downloading arduino:dfu-util@0.10.0-arduino1
downloading arduino:rp2040tools@1.0.2
downloading arduino:mbed_nano@2.0.0
Failed to install platform: arduino:mbed_nano.
Error: 2 UNKNOWN: installing tool arduino:arm-none-eabi-gcc@7-2017q4: moving extracted archive to destination dir: rename C:\Users\Mike\AppData\Local\Arduino15\tmp\package-827321993\gcc-arm-none-eabi-7-2017-q4-major-win32 C:\Users\Mike\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4: Access is denied.`

Cecchellone, 2021-05-12:
same error but with `esp8266:esp8266@2.7.4` platform

Cecchellone, 2021-05-12:
Partially solved by running in powershell:
```
cd C:\Users\<UserName>\AppData\Local\Arduino15
icacls .\packages\* /grant Everyone:F /r
```

makerinchina-iot, 2021-07-18:
> Partially solved by running in powershell:
> 
> ```
> cd C:\Users\<UserName>\AppData\Local\Arduino15
> icacls .\packages\* /grant Everyone:F /r
> ```

the same result, and use the powershell cmd it just shows: "invalid parameter '/r'. solved by this cmd:
`icacls.exe .\*  /grant[:r] Everyone:F`