arduino/arduino-ide #319: IDE-Editor  Beginner or Advance mode

trlafleur, 2021-04-15:
With the capabilities of the new IDE, having and beginner and advance mode would be very useful to advance programmers...

Allowing for the advance function to be available in the sidebar or as an option in a user-defined drop-down box, or pallet.

mink99, 2021-04-17:
maybe it would help to get rid of the "vi" like interface (F1 and select/type command) and provide proper menu items for standard tasks like toggle outline toggle explorer etc. come on, VI is from 1976.... 
btw this is the reason i have dropped vscode.

runger1101001, 2021-04-27:
Yeah, what happened? Pro-IDE had an "Advanced Mode" before it was retired... any chance of seeing a comeback?

While I understand the desire to keep the barriers to entry low, it would also be cool to have a more standard vscode view with file-browser, git functionality, etc... or am I just missing where this is hidden?