arduino/arduino-ide #321: In Dark mode top of the window is white

ednieuw, 2021-04-17:
**Describe the bug**
In dark mode the top of the windows is white

**Expected behavior**
This should also be black

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows10
 - Version: 2.0.0 beta 4

ClaudioCas, 2021-04-17:
Probably is not really a bug, but a dark header like VS Code will be really great!
[image]

kittaakos, 2023-04-27:
> but a dark header like VS Code will be really great!

The dark window title bar in VS Code on Windows is not native but custom. It's handled by the `window.titleBarStyle` setting, currently disabled on Windows and not supported by Theia IDE on macOS: https://github.com/arduino/arduino-ide/pull/1870

closed 2023-05-09 by kittaakos