arduino/arduino-ide #322: Problems with port detection with Raspberry Pico

sstaub, 2021-04-17:
Using Raspberry Pico with official Arduino Mbed Core

When plugging in my Pico the Port is automatic detected and selected, but the port is twice in the list
[image]
After uploading, the port is not automatic detected and I have to select the port manually.
[image]

This behavior is different to classic IDE where the automatic port detection after upload works as expected.

**Desktop (please complete the following information):**
 - OS: macOS 11.2.3
 - Version: 2.0.0 beta 5

g2721, 2021-07-23:
Is this problem still there?

sstaub, 2021-07-24:
The problem seems solved with Build 9.

closed 2021-07-24 by sstaub