arduino/arduino-ide #324: Splash Screen Should not be Always on Top

Logicer16, 2021-04-18:
**Describe the bug**
When launching the application, the splash screen will force itself on top of all other applications stop interaction with other applications.

**To Reproduce**

1. Start application
2. Wait for splash screen
3. Attempt to make another window active

**Expected behavior**
A user would expect to be able to use their computer without having to wait for the application to start, as this may take some time (especially on slower devices). The splash screen should continue to show on top once the application is launched but should not continue stay on top of applications made active after the splash screen is shown.

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Education & macOS Big Sur 11.2.3
 - Version: 2.0.0-beta 5

Logicer16, 2021-04-18:
Verified on 2.0.0-beta.5-nightly.20210418 (latest) for both platforms.

sterretjeToo, 2021-12-21:
I've done very limited playing with the beta releases but I'm planning to make RC2 my default IDE (currently running the zip version.

I would like to add a vote that this is extremely annoying behaviour; I think that this has an easy fix (I might be wrong) and hence would have expected this to be fixed in release candidates already.

intg76, 2022-07-26:
it's so easy to fix, why it don't fixed yet?

Logicer16, 2022-07-26:
Unless we're going to end up removing the splashscreen, the pr to fix this issue is already open in #717 (see [comment](https://github.com/arduino/arduino-ide/pull/851#issuecomment-1057714765) in #851 as to why that one is preferable). All that needs to be done now is to merge the pr.

commit 334b52be (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #324
Closes #327
Closes #717
Closes #851
commit ed97fc84 (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #324
Closes #327
Closes #717
Closes #851
commit 0e4a6a6c (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
commit f7a13539 (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
closed 2022-08-09 by kittaakos