arduino/arduino-ide #326: Offer ESP8266, ESP32, nRF, STM,... boards by default

probonopd, 2021-04-18:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to fiddle around with json URLs and whatnot just to get an ESP8266 (e.g., NodeMCU), ESP32, nRF, or STM (e.g., "Blue Pill" STM32F103C8T6) board up and running.

**Describe the solution you'd like**

ESP8266 (e.g., NodeMCU), ESP32, STM board up (e.g., "Blue Pill" STM32F103C8T6) boards should be listed __by default__ here:

[image]

**Describe alternatives you've considered**
Fiddling around with json URLs and whatnot 

**Additional context**
* ESP8266, ESP32, nRF, and STM boards are arguably among the most popular development boards in the world
* Basically, you "just" need to integrate these right out of the box:
  * https://github.com/esp8266/Arduino
  * https://github.com/espressif/arduino-esp32
  * https://github.com/sandeepmistry/arduino-nRF5
  * https://github.com/stm32duino/Arduino_Core_STM32
* Maybe collaborate with/reuse the infrastructure of PlatformIO

probonopd, 2021-04-18:
These URLs should be added __by default__ to the board manager. It is not necessary to bundle the cores with the IDE. It is OK to bring up a big red warning if the user selects one of those, saying "This board is not supported by arduino.cc and is offered by a third party". 

```
https://arduino.esp8266.com/stable/package_esp8266com_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
```

probonopd, 2021-04-18:
What I would like to see is that out of the box (without the user having to do anything), `~.arduinoIDE/arduino-cli.yaml` looks like this:

```
board_manager:
  additional_urls:
  - https://arduino.esp8266.com/stable/package_esp8266com_index.json
  - https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
  - https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
  - https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
daemon:
  port: "50051"
directories:
  data: /root/.arduino15
  downloads: /root/.arduino15/staging
  user: /root/Arduino
ide:
  bundled: false
  portable: false
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
sketch:
  always_export_binaries: false
```

per1234, 2021-04-18:
Hi @probonopd. Thanks for taking the time to make a feature request.
Arduino has already made it extremely easy to install support for 3rd party boards platforms. It's no great hardship to add a URL to your preferences.

closed 2021-04-18 by per1234

probonopd, 2021-04-18:
@per1234 thanks for your quick response but I'd like you to consider your response again for a minute.

It is a great hardship. It is more hassle than needed. And worst of all, it sends the totally wrong signal to the open source community.

What is outright _offending_ to the open source community is that commercial boards show up there, including this

[image]

and this

[image]

yet the most popular ones used in the community are missing!
I would almost bet that the above combined don't have a fraction of users compared to the ones I am suggesting.

At this point, the Arduino IDE feels like a commercial ad space to me. Is Arduino.cc selling these spaces or money or what?

per1234, 2021-04-18:
Arduino has invested a huge amount of effort and money into making this free open source software support 3rd party hardware that directly competes with their products. This, even after many (with some very notable exceptions) of the companies who benefit hugely from that support have made no real efforts to contribute to this project.