arduino/arduino-ide #327: Application start up is indefinite when window is minimised

Logicer16, 2021-04-18:
**Describe the bug**
If the loading project Windows is minimised while the application is starting the starting process will not continue and the splash screen will stay visible indefinitely (or at least for an abnormally longer amount of time).

**To Reproduce**
1. Start application
2. Minimise the loading project window
3. Loading will not continue and splash screen stays visible 

**Expected behavior**
Starting will continue or the splash screen will be minimised along with the window

**Screenshots**
[image]
(Stalls at splash screen for some time)

**Desktop (please complete the following information):**
 - OS: macOS Big Sur 11.2.3
 - Version: 2.0.0-beta.5-nightly.20210418

Silver-Fang, 2022-07-26:
OK, so this issue is not fixed for more than a year …

Logicer16, 2022-07-26:
Even though it won't solve the issue, #717 may make this issue far less noticeable

Silver-Fang, 2022-07-26:
> Even though it won't solve the issue, #717 may make this issue far less noticeable

So this is an extremely hard-to-fix iceberg bug that involves the design of the underlying architecture?

Logicer16, 2022-07-26:
> So this is an extremely hard-to-fix iceberg bug that involves the design of the underlying architecture?

I wouldn't say that. I'm not sure what it's causing this, I was just pointing out that the pr I mentioned would make it less noticeable as it allows other windows to be placed on top of the splashscreen

commit 334b52be (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #324
Closes #327
Closes #717
Closes #851
commit ed97fc84 (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #324
Closes #327
Closes #717
Closes #851
commit 0e4a6a6c (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
commit f7a13539 (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
closed 2022-08-09 by kittaakos