arduino/arduino-ide #333: Add an "upload binary to board" option in the IDE

jerabaul29, 2021-04-20:
There is a nice feature under:

```
Sketch > Export compiled Binary
```

which allows to dump the binary of the sketch under a .bin file.

Would it be possible to also provide an "Upload compile Binary" option somewhere in the IDE, so that the users can upload their pre compiled binaries from the IDE, without the need to use yet another tool?

per1234, 2021-04-26:
Hi @jerabaul29. Thanks for taking the time to submit this feature request.

You'll be glad to learn that this functionality is already provided by `arduino-cli upload --input-file`:
https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_upload/#options
Arduino CLI is part of the Arduino IDE 2.x installation, but you might find it more convenient to just install a standalone copy of Arduino CLI for using directly:
https://arduino.github.io/arduino-cli/latest/installation/

Since uploading a pre-compiled binary is something only for advanced users, there is no need to provide a capability to do it via a GUI.

closed 2021-04-26 by per1234

jerabaul29, 2021-04-26:
Thanks! :)

You are the people developing the IDE and it is natural that you decide how you want it to look like, but honestly, I do not think that this is such an advanced feature. I know of some colleagues who have 0 experience with Arduino and needed to upload some binaries on a board lately. Since this was not directly available into the IDE, they used a third party software / GUI for doing that, which took them time and was a big buggy process. I think that there are quite many cases in which beginners would like to be upload just by pressing a button which would make their life much easier than finding out of how to use a command line method.

Jens869, 2022-09-29:
Version: 2.0.0
Datum: 2022-09-14T07:08:28.861Z
CLI Version: 0.27.1 [a900cfb2]

Sketch > Export compiled Binary

Mac OS

the bin file is found in:  /private/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino-sketch-352F5EAEA50A9FC581E2424D67B8F822/VFD-8266-Uhr.ino.bin
but not in:  /Users/<username>/Documents/Arduino/VFD-8266-Uhr/VFD-8266-Uhr.ino.bin

the subfolder is hard to reach

But it old Arduino IDE 1.x puts it in the right folder !

Jabadabado, 2024-09-29:
I too and many on the internet are having this issue with loading a hex or bin file directly from the Arduino IDE. Doing this via CLI only, is scaring away many customers.