arduino/arduino-ide #337: Format Document not works for me

JAICHANGPARK, 2021-04-22:
**Describe the bug**
- I'd like to auto format my code 
- so I tried to format document 
- but It's not working for me

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10 ]
 - Version: [e.g. 2.0.0-beta5]

Theagainmen, 2021-04-27:
Same here, came to make an issue but see it's already an open one :). 

[image]

JAICHANGPARK, 2021-05-24:
Any have update this issue?

closed 2022-06-09 by per1234