arduino/arduino-ide #339: USBasp gone?!

kr15uk, 2021-04-24:
Flash using USBasp option seems to be gone, is it?!
Thanks.

 - OS: macOS BigSur ARM
 - Version: 2.0 beta5

per1234, 2021-04-24:
Hi @kr15uk. USBasp is still supported. A change (https://github.com/arduino/Arduino/pull/9900) was made in Arduino IDE 1.8.13, which persists into Arduino IDE 2.x as well, which makes it so only the programmers[ defined](https://arduino.github.io/arduino-cli/latest/platform-specification/#programmerstxt) by the platform of the currently selected board (and the platform of its core in case it uses a [core reference](https://arduino.github.io/arduino-cli/latest/platform-specification/#core-reference)).

So if you have selected a board that doesn't define a USBasp programmer, or you have no board at all selected, then you won't see USBasp listed under the **Tools > Programmer** menu.

If you have selected one of the boards of the Arduino AVR Boards platform, then you should definitely see a **Tools > Programmer > USBasp** item. If you have selected a board that is not compatible with the USBasp, then it is normal and expected for it to not be listed in that menu, since it would not be of any use. It's possible there are some 3rd party AVR boards that have not adapted to the change by providing their own USBasp definitions. I'm not aware of any such, but if there are that would be a matter to take up with the maintainer of the platform after verifying they haven't already done so.

Please try this:
1. **Tools > Board > Boards Manager**
2. Verify that the "Arduino AVR Boards" platform is shown as "INSTALLED".
3. **Tools > Board > Arduino AVR Boards > Arduino Uno**
4. Open the **Tools > Programmer** menu.

Do you see "USBasp" listed in the menu there?

kr15uk, 2021-04-24:
Perfect, all sorted now.
Had to install AVR support from the board manager.
Funny though it had ESP8826, ESP32, Due etc already preinstalled but not AVR.
Just assumed AVR is there by default (hence others were) and new v2.0 does some funky detection like Platform IO/AVRfuses does.
Oh well, should have been more thorough I guess.

Thanks for your help.

closed 2021-04-24 by per1234