arduino/arduino-ide #341: Failed to start language server for esp32:esp32:esp32s2

shumifan16, 2021-04-27:
After installing Arduino IDE 2.0 and the ESP32 boards I get the error in the cli window I started the IDE from.
Ubuntu 16.04
Arduino IDE 2.0
Board selected ESP32S2 Dev Module

Messages:
root INFO [fa769db7-69bd-45d3-b030-014c68cdc75b] Start of 5 plugins took: 1486.5 ms
root INFO Starting language server: esp32:esp32:esp32s2
root INFO Failed to start language server for esp32:esp32:esp32s2
root INFO Starting language server: esp32:esp32:esp32s2
root INFO Failed to start language server for esp32:esp32:esp32s2

All still seems to work. I can compile and upload the code to the chip, just wondering what is missing.

koendv, 2021-04-30:
I'm having a similar issue:
```
root INFO Starting language server: STMicroelectronics:stm32:GenF1
root INFO [hosted-plugin: 3690] PLUGIN_HOST(3690): PluginManagerExtImpl/loadPlugin(/home/pi/opt/arduino-ide/resources/app/plugins/vscode-arduino-tools/extension/dist/bundle)
root INFO Failed to start language server for STMicroelectronics:stm32:GenF1
root INFO Starting language server: STMicroelectronics:stm32:GenF1
root INFO Failed to start language server for STMicroelectronics:stm32:GenF1
root INFO Starting language server: STMicroelectronics:stm32:GenF1
root INFO Failed to start language server for STMicroelectronics:stm32:GenF1
root INFO Starting language server: STMicroelectronics:stm32:GenF1
root INFO Failed to start language server for STMicroelectronics:stm32:GenF1
```
Compiling and uploading is fine.

joysfera, 2021-11-03:
With up-to-date versions of everything the language server seem to start OK, but for ESP32 boards it does not work (though it does work for ESP8266 and other Arduino boards). No obvious error message, no log found (even though it is enabled in properties):

```
2021-11-03 12:12:57 root INFO Board config changed: 
2021-11-03 12:12:57 root INFO Starting language server: esp32:esp32:pico32
2021-11-03 12:14:18 root INFO Board config changed: 
2021-11-03 12:14:18 root INFO Starting language server: esp8266:esp8266:espino
2021-11-03 12:15:06 root INFO Board config changed: 
2021-11-03 12:15:06 root INFO Starting language server: esp32:esp32:esp32wrover
```

per1234, 2022-03-11:
Closing due to lack of response.

I tried it with both Ubuntu 20.04 (x86-64 host) and Windows 10, using Arduino IDE 2.0.0-rc5 and the `esp32:esp32@2.0.2` boards platform. I only get the expected command line output:

```text
root INFO Starting language server: esp32:esp32:esp32s2
```
(no failures)

@shumifan16 or @koendv if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-03-11 by per1234