arduino/arduino-ide #343: Show in the Tools dropdown menu the chosen options

Theagainmen, 2021-04-27:
**Is your feature request related to a problem? Please describe.**
A board has many settings, when you click the `Tools` dropdown menu a user would like to see all the options in one go. Now the user has to manually holds it mouse over each option to see what's selected.

**Describe the solution you'd like**
The same as in Arduino IDE 1.x where it would always show the selected option.

**Describe alternatives you've considered**
None.

**Additional context**
Arduino IDE 1.x the `Tools` dropdown menu looks like this:
[image]

Arduino IDE 2.0 it looks like this:
[image]
All the possible options looks hidden this way, and you have to manually check all of them. My upload speed was wrong for example, and usually you notice that in one go when you hover your mouse over 'tools'. But now you specifically need to hover over all of them to double check these settings :). 

If you have any questions or comments, please let me know.

Lorandil, 2021-04-27:
That bothers me, too :)

DaleMitchell, 2022-10-01:
Are there any plans to add this back? @per1234  ?

DaleMitchell, 2022-10-01:
Are there any plans to add this back? @per1234   ? @AlbyIanna  ?

I very much miss this feature and share the frustration of others that this was removed in Arduino IDE 2.0 for unknown reasons.

This feature allows for easy documentation of options selected for the board, a single screenshot can be taken. Also, it allows for double checking each option at a quick glance without having to tediously hover over each option.

PR #1746 (2022-12-08): feat: show the selected board config value on menu
### Motivation

As per request from #343, IDE2 shows the selected board config value on the menu item.

https://user-images.githubusercontent.com/1405703/206217908-c99d066a-cad9-4c6a-867c-7150554439e9.mp4

### Change description

Append the selected board config value to the menu label.

### Other information

Closes #343

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and descr [... cut]
commit af468a73 (2022-12-08): feat: show the selected board config value on menu

closes #343
closed 2022-12-08 by kittaakos