arduino/arduino-ide #347: Not compiling with #include <Ethernet.h> [Compilation error: Error: 2 UNKNOWN: exit status 1]

Mr-Zero88, 2021-04-30:
**Describe the bug**
when i include the Ethernet library for the EthernetClient it won´t compile and says "Compilation error: Error: 2 UNKNOWN: exit status 1" wenn i compile the same code with the older ide(1.8.13) it works.
And next to that it also removes thhe type definition of byte.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project.
2.  Include the Ethernet library.
3. Create a byte.
4. Compile.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: 20H2(19042.928)

per1234, 2021-04-30:
Hi @Mr-Zero88. Thanks for taking the time to submit a bug report.

I'm going to ask you to post some additional information that might help us to identify the problem.
Please do this:

Please do this:

1. Select **File > Preferences** from the Arduino IDE's menus.
1. Check the box next to **"Show verbose output during: [ ] compilation"**.
1. Click the **<kbd>OK</kbd>** button.
1. Select **Sketch > Verify/Compile** from the Arduino IDE's menus.
1. Wait for the compilation to fail.
1. Right click on the black "Output" pane at the bottom of the Arduino IDE 2.x window.
1. From the context menu, click **Copy All**. This will copy the full error output to the clipboard.
1. Paste the copied output in a reply here.

Mr-Zero88, 2021-05-03:
`Using board 'controllino_maxi' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\CONTROLLINO_Boards\hardware\avr\3.1.0
Using core 'arduino' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\CONTROLLINO_Boards\\hardware\\avr\\3.1.0\\variants\\Controllino_maxi" "C:\\Users\\User\\AppData\\Local\\Temp\\arduino-sketch-94E9CE59DA540845567ED8079F0B5BAE\\sketch\\buttons.ino.cpp" -o nul
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\CONTROLLINO_Boards\\hardware\\avr\\3.1.0\\variants\\Controllino_maxi" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\libraries\\SPI\\src" "C:\\Users\\User\\AppData\\Local\\Temp\\arduino-sketch-94E9CE59DA540845567ED8079F0B5BAE\\sketch\\buttons.ino.cpp" -o nul
Alternatives for Controllino.h: [CONTROLLINO@3.0.7]
ResolveLibrary(Controllino.h)
  -> candidates: [CONTROLLINO@3.0.7]
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\CONTROLLINO_Boards\\hardware\\avr\\3.1.0\\variants\\Controllino_maxi" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\libraries\\SPI\\src" "-Ic:\\Users\\User\\Documents\\Arduino\\libraries\\CONTROLLINO" "C:\\Users\\User\\AppData\\Local\\Temp\\arduino-sketch-94E9CE59DA540845567ED8079F0B5BAE\\sketch\\buttons.ino.cpp" -o nul
Alternatives for LinkedList.h: [LinkedList@1.3.2]
ResolveLibrary(LinkedList.h)
  -> candidates: [LinkedList@1.3.2]
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\CONTROLLINO_Boards\\hardware\\avr\\3.1.0\\variants\\Controllino_maxi" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\libraries\\SPI\\src" "-Ic:\\Users\\User\\Documents\\Arduino\\libraries\\CONTROLLINO" "-Ic:\\Users\\User\\Documents\\Arduino\\libraries\\LinkedList" "C:\\Users\\User\\AppData\\Local\\Temp\\arduino-sketch-94E9CE59DA540845567ED8079F0B5BAE\\sketch\\buttons.ino.cpp" -o nul
Alternatives for Ethernet.h: []
ResolveLibrary(Ethernet.h)
  -> candidates: []
Using library SPI at version 1.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries\SPI 
Using library CONTROLLINO at version 3.0.7 in folder: c:\Users\User\Documents\Arduino\libraries\CONTROLLINO 
Using library LinkedList at version 1.3.2 in folder: c:\Users\User\Documents\Arduino\libraries\LinkedList 
Compilation error: Error: 2 UNKNOWN: exit status 1`

per1234, 2021-05-03:
OK, that is helpful. 

The classic Arduino IDE comes with some of the most popular official Arduino libraries pre-installed. So you might be accustomed to having the Ethernet library always available. Unlike the classic Arduino IDE, Arduino IDE 2.x does not currently come with any libraries pre-installed. So you must install all the libraries you need via Library Manager.

You can install the Ethernet library by following these instructions:
1. Select **Sketch > Include Library > Manage Libraries** from the Arduino IDE's menus. This will open Library Manager in the IDE's side bar on the left side of the window.
2. In the Library Manager's **"Filter your search..."** field, type `ethernet`.
3. Scroll down through the list of search results until you find "Ethernet by Various". Click on it.
4. Click the **<kbd>INSTALL</kbd>** button.
5. Wait for the installation to finish.

You should now be able to compile your sketch.

Since this didn't end up being a bug in the Arduino IDE, I'm going to go ahead and close this. If you have any questions, I'm always available to help out over on the Arduino Forum's dedicated category for Arduino IDE 2.x:
https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93

closed 2021-05-03 by per1234