arduino/arduino-ide #348: serial monitor: missing send button / not responding to enter

ShadowRStar, 2021-05-01:
**Describe the bug**
To me it appears as there is no send button in the serial monitor. furthermore afai can tell the sm doesnt respond to enter. 

**To Reproduce**
Steps to reproduce the behavior:
1. open sm
2. no send btn
3. no way to send with enter as well
**Expected behavior**
Add a send button and make the 
**Screenshots**
[image]
[image]

 - OSX 10.15.7
 - Version: nightly 20210501

per1234, 2023-02-21:
I will reopen this since the send button has not been added yet. We will close it once that work is done.

Thanks for your report **strau0106**!

strau0106, 2023-02-21:
Ok, just thought it was ready to close since there was no activity

per1234, 2023-02-21:
I think this will be done as part of a complete rework of Serial Monitor to be a dedicated window: https://github.com/arduino/arduino-ide/issues/289. This means it is dependent on a significant amount of other work than would be the case if the button was only added to the current provisional Serial Monitor component.

The new design will be aligned with the existing Serial Plotter feature, which does have a send button consistent with the Arduino IDE 1.x interface:

[image]