arduino/arduino-ide #350: Ardublocks needed for teaching

loessi, 2021-05-02:
I would appreciated the further possibilty to use ARDUBLOCK for teaching our children at school.

When do You see the time to make it possible.

Thanks Reinhard

ubidefeo, 2021-05-02:
hi @loessi 
Ardublock is not a project by Arduino and is developed in Java, which does not make it possible to integrate easily in Arduino IDE 2.0

If the community is interested someone can build it in JS+HTML and build a Theia Extension for the Arduino IDE but we do not plan to build it ourselves.
Our team is too small to undertake such big tasks

per1234, 2021-05-02:
I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/58.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-05-02 by per1234