arduino/arduino-ide #351: IDE startup takes too long when a soft link into big directory structures exist in the "sketchbook location" directory

Ho-Ro, 2021-05-03:
**Describe the bug**
The IDE startup takes forever when links into complex directories with thousands of subdirectories exist in "sketchbook location" directory structure. 

**To Reproduce**
Steps to reproduce the behavior:

1. Go to the sketchbook storage directory (in my case `cd ~/Arduino/Projects`).
2. Create a link into a complex directory structure with circular links.
3. Start the ide (beta5).
4. The IDE opens, but the splash screen is displayed forever, warnings are shown: 
````
root WARN Could not load sketch from ...
root WARN Could not load sketch from ... 
...
root WARN Could not load sketch from ...
root WARN Frontend EditorNavigationContribution.onStart is slow, took: 450821.8 ms
````
5. Startup takes almost 8 minutes.
5. After removing the critical link the IDE starts much faster.

**Expected behavior**
The splash screen closes fast, and the IDE works normally - independent of the project directory structure.

**Desktop**
 - OS: debian stable with KDE
 - Version: arduino-ide beta 5

willie68, 2023-05-11:
Is there any improvment in sight?
Todays measurement: (i9, 32GB, M2 SSD) 
2.1.0 time to get the first sight after start: 45 sec, time until i can code: 1:45
1.8.19: time to get the first sight after start:  6 sec, time until i can code: 0:45
Both with the same workspace/sketchbook.

and for fun: VS Code (with activated support of JS, Java and Golang, all in one project)
time to get the first sight after start:  3 sec, time until i can code: 0:15

It's even getting worther than ever...

Ho-Ro, 2023-05-11:
> because it's not the way you're supposed to work in the Arduino context

But it is my way to organize my work, e.g. symlink to a doc directory with datasheets that point to some papers in another section that points to the datasheets.
One of this circles is enough to confuse the 2.0 while 1.x is fine...

ubidefeo, 2023-05-11:
Is the symlink inside your sketch or in your sketchbook folder?

willie68, 2023-05-11:
I don't think symbol links are the problem. I have 65 folder in the sketchbook, some with subfolder for different projects. (Mainly because many projects live in git repos) I think round about 200 projects. But the libraries folder is crowded. 265 libs, but mostly named arduino-xxxxxx which are older versions of the libs. And that will cause the problem. The ide has to scan every libs folder for the infos of keywords and so on. And the lib manager lacks the function of deleting older version. And its a mess when you tried to do it manually. The other part is, scanning the folder seems to be a syncron task. If you implement that to async this would help much. That's how professional IDE do such scanning.

ubidefeo, 2023-05-11:
oh, now I get it
legacy libraries need a lot more work, that is a known problem.
scanning the libraries is also required to populate the examples menu, and that's something that professionals IDEs don't need to do.
This IDE is targeted to Arduino projects and we have a lot of legacy to respect in order not to break things.
That said, optimising it and making it better is a clear goal of ours :)

Our backlog is large, and the team is small...
Bear with us 🙏🏼

willie68, 2023-05-12:
I'm willing to help, I'm a pro, but my problem is that one has switched to Elektron and JS. That's not my preferred language. I'm more of a backend guy, Java, Golang, C++...
I'm just doing a few smaller things in JS, little Vue sites and stuff like that, but no bigger projects.

Ho-Ro, 2023-05-12:
For me the 2.1.0 improves the behaviour, startup time until I can start working is 15 seconds - faster than 2.0.0 and much faster than the .rc versions.

ok for me -> close

Thx to the whole team for your effort!

willie68, 2023-07-15:
Unfortunately I have to disappoint you, the current 2.1.1 now needs 4:30min for the 1st start, with my main project folder. That's a whopping 50% more than the last 2.0
And then the version doesn't even get along with the folder structure that version 2.0 left behind. See issue #2141

willie68, 2023-07-15:
https://www.dropbox.com/s/3lfm23biapssr46/arduino_2.0_startup.mp4?dl=0

This little screencam video is just for showing the problem:

first i started my vs code ide with PlatformIO for arduino programming.
Starting at 1sec, i can start programming after 18 secs.  after 2 sec you will see the vs code window to appear.

Than starting Arduino IDE 2.1.1 with an nearly empty sketchbook at 30sec at 53sec the first sight of the ide is coming. in this 23sec you will see nothing... thats worth! start coding than at 1:03 -> that are 33sec for an nearly empty sketchbook.
Than i switch to the old productive folder with all my needed libs...
Starting at 1:52 (now the first 23sec will be shrunked, because most of the files are now in the os cache) 
ready to start programmin at 5:30 -> 3:08

My PC is an lenovo yoga slim 7 with an i7 10750H, 16GB RAM and a fast m2 SSD.

per1234, 2023-07-15:
@willie68 is there a soft link into a big directory structure under your sketchbook folder (the path configured via the "**Sketchbook location**" IDE preference)?

@Ho-Ro do find that there has been a regression in the IDE in regards to this issue since the time of your previous report that the problem was resolved?

willie68, 2023-07-16:
Hi,
no, no softlinks in the sketchbook. Just simple folders.

My problem is not that the IDE needs time to read the folder structure.
My biggest problem is that it takes a long time before the IDE even shows any sign of life. Unfortunately, this means that several attempts are made to start. (These are my own experiences from my Arduino workshops)
Then you suddenly have 3 or 4 IDEs open.
The other thing is that the analysis blocks the actual IDE. This could very well be done in parallel. (See VS code)
Then the user could already edit the files, maybe not compile them yet, but at least the IDE would be responsive.

Ho-Ro, 2023-07-16:
> o find that there has been a regression in the IDE in regards to this issue since the time of your previous report that the problem was resolved?

I tested again after a fresh boot of my system and it took again a very long time. The next try was much faster due to the already buffered disk content. Version 1.x is still faster and more responsive.