arduino/arduino-ide #357: Compiler output not displayed

carlosperate, 2021-05-07:
**Describe the bug**
This sketch from the BLE test dashboard has some issues with the latest version of the MbedOS core: 
- https://arduino.github.io/ArduinoAI/BLESense-test-dashboard/
- https://create.arduino.cc/editor/dpajak/e4128780-d76d-438f-97a4-541740cb0473/preview

When trying to compile it with IDE 2.0.0-beta.5 the compilation error is not shown in the output.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Nano 33 BLE board (MbedOS)
2. Create a new sketch with this code sketch
3. Select the Nano 33 BLE board type
4. Click "Verify" button
5. Output only shows the compiler error code, not the full message

**Expected behavior**
Full compiler output, specifically the errors.

**Screenshots**
beta.5:
[image]

Old IDE:
[image]

**Desktop (please complete the following information):**
 - OS: macOS Big Sur
 - Version: 11.2.1

**Additional context**
N/A

petrbrouzda, 2021-05-15:
Windows 10, IDE 2.0.0-beta6.
The same problem - **stderr from compiller is not displayed,** so I'm not able to solve problems. Have to build app in Arduino 1.8 to see root cause.

See image with 1.8 and 2.0-beta6 difference on the same source:
[image]

per1234, 2021-05-16:
Thanks to @carlosperate for taking the time to submit a report and and to @petrbrouzda for pointing out the previous related issue. This also might be related to https://github.com/arduino/arduino-ide/issues/374

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-05-16 by per1234