arduino/arduino-ide #358: Editor Quick Suggestions isn't Presistant in IDE 2.0 Beta 5

4dbcc7e6c2, 2021-05-08:
On startup, the Editor Quick suggestions isn't enabled.  Then after enabling it in preferences, it doesn't activate.  When you go back into preferences, it is uncheck.  This was the same with Beta 4, however at least it held the setting whilst the program was open.  

if more detail/logs are needed, I am more than happy to supply

Hope this can be fixed as 2.0 is FAR FAR better than VScode or 1.8

Nick-Norton, 2021-05-15:
IDE 2.0 Beta 5 I am having the same issue. Same issue also with Version: 2.0.0-beta.6
I am using the Linux build.
The check box state for Editor Quick Suggestions is not retained.
[image]

per1234, 2021-06-13:
Thanks for taking the time to submit an issue @4dbcc7e6c2 .

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/313.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-06-13 by per1234