arduino/arduino-ide #359: libnss3.so: cannot open shared object file

maxiauer, 2021-05-08:
1) Created a fresh Linux Developer environment on ChromeOS 90.0.4430.100. ChromeOS uses Debian Buster

2) Download and Installed Adruino IDE 1.8.13 - Works OK, no issue

3) Download Adruino IDE 2.0 Beta 5 - According to the Install doc, I should run ./arduino-ide and it should work... but it doesn't

max@penguin:~/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit$ ./arduino-ide
./arduino-ide: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory
max@penguin:~/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit$ sudo ./arduino-ide
./arduino-ide: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory

Seems some file is missing, but I'd believe that the idea is the downloaded file would contain all files/libraries required, like it does for IDE 1.8, without needing to install extra stuff. Or if something extra is needed, the documentation would indicate that.

Thanks !!!

kittaakos, 2021-05-10:
> 1. environment on ChromeOS 90.0.4430.100

> 3\. it should work... but it doesn't

I'm afraid, it won't work: https://github.com/electron/electron/issues/12712

kittaakos, 2021-05-10:
Here is the upstream Theia issue: https://github.com/eclipse-theia/theia/issues/4725

kittaakos, 2021-05-10:
Does it help if you install `libnss3-dev` manually? (ref: https://github.com/symfony/panther/issues/7#issuecomment-411446756)

maxiauer, 2021-05-11:
Thanks

So, it helped... After manually installing the files:

_The following NEW packages will be installed:
  libnspr4 libnspr4-dev libnss3 libnss3-dev_

Now the app starts and I get a GUI with the same project I was working with last time I used 1.8.13

I do not have the time to fully test this until the weekend (create something, compile and upload), but the logs shown on the console are not that great... not sure if I shall worry about the following 2 errors (or report them on a new issue)

_root ERROR Error: libxkbfile.so.1: cannot open shared object file: No such file or directory
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at NativeBinding._init (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/native-keymap/index.js:15:22)
    at NativeBinding.getCurrentKeyboardLayout (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/native-keymap/index.js:28:10)

root ERROR TypeError: Cannot read property 'getKeyMap' of null
    at NativeBinding.getKeyMap (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/native-keymap/index.js:20:29)
    at Object.exports.getKeyMap (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/native-keymap/index.js:58:18)
    at ElectronKeyboardLayoutProvider.getNativeLayoutSync (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/electron-node/keyboard/electron-keyboard-layout-provider.js:36:35)
    at ElectronKeyboardLayoutProvider.getNativeLayout (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/electron-node/keyboard/electron-keyboard-layout-provider.js:31:37)
    at JsonRpcProxyFactory.<anonymous> (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:213:73)
    at step (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:48:23)
    at Object.next (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:29:53)
    at /home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:23:71
    at new Promise (<anonymous>)
    at __awaiter (/home/max/Programs/arduino-ide_2.0.0-beta.5_Linux_64bit/resources/app/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:19:12)_

Will report back after the weekend
Thanks

kittaakos, 2021-05-11:
OK, I did not know [this](https://github.com/Microsoft/vscode/issues/1031#issuecomment-533717495): many Chromebooks come with an ARM chip. VS Code had the same issue till last year. Related: https://github.com/arduino/arduino-ide/issues/107

> not sure if I shall worry

If your Chromebook has an ARM chip, I would expect some issues with the language features too.

maxiauer, 2021-05-16:
Back again, with the requested info:

**1) System Info**

_max@penguin:~$ cat /etc/*-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

max@penguin:~$ uname -a
Linux penguin 5.4.99-12983-g8b7876ab9f5e #1 SMP PREEMPT Sun Feb 21 11:49:12 PST 2021 x86_64 GNU/Linux_

By the way the platform is Intel Core i7 (most of Chromebooks running Linux VM/Container are, although I think they have launched the feature also in a few ARM based ones)

**2) Theia Electron**. Runs OK on the system (remember I installed libnss3-dev). I saw a few errors and warnings, mainly the same ones I saw on the console of the Arduino IDE. Will put them in another issue to avoid mixing problems here. The Keyboard issue seems related that Chromebooks have different special keys (as an example, they lack the WIN key)

**4) Script and Fresh Install**

I deleted the Crostini Container (the Debian VM) and aked ChromeOs to create a new one to start from scratch before running your script. I did the usual update/upgrade and just in case checked that nothing changed on the kernel (run uname again). I had to change the repository of your script at it seems https://github.com/bcmi-labs/arduino-editor.git does not exist anymore (I used https://github.com/arduino/arduino-ide.git)

_...
$ theia rebuild:electron
Processing @theia/node-pty
Processing nsfw
Processing native-keymap
Processing find-git-repositories
Processing drivelist
✔ Rebuild Complete
Done in 9.24s.
yarn run v1.22.5
$ theia start --plugins=local-dir:../plugins
/home/max/dev/git/arduino-editor/node_modules/electron/dist/electron: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory
Done in 0.36s._

I checked all output looking for other errors about missing libnss3.so, but couldn't find any... The script "runs" but it seems it simply assumes that the libnss3.so will be there... (either as part of one of the modules loaded or as part of the OS)

INTERESTINGLY I tried to run the Theia GUI again and...

_max@penguin:~$ ./TheiaBlueprint.AppImage 
/tmp/.mount_TheiaBm5zT5r/theia-blueprint: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory_

A Google search shows several apps where ChromeOS Linux users are complaining about missing libnss3.so

Everything solved again after manually _sudo apt install libnss3_ Theia GUI and your script works OK

No sure what is the package that contains libnss3.so that "everybody seems to have installed on their systems"... In any case it would be a good idea to check for it and alert the user (or proactively include it on the package just in case)

kittaakos, 2021-05-26:
I opened a generic issue here: https://github.com/eclipse-theia/theia-blueprint/issues/100
Hopefully, the Theia community has an idea on how to solve this issue at build time.

> after manually _sudo apt install libnss3_ Theia GUI and your script works OK

I am glad you found a workaround.

wired-filipino-owl, 2023-03-05:
I'm also working off of a Chromebook.
In addition to installing `libnss3-dev`, I also had to install the package `libsecret-1-0` to get the Arduino IDE to launch. Installing that package gets rid of this error:
```
Error: libsecret-1.so.0: cannot open shared object file: No such file or directory
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:5:1800)
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1170:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:5:1800)
    at Module.load (node:internal/modules/cjs/loader:982:32)
    at Module._load (node:internal/modules/cjs/loader:823:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1006:19)
    at require (node:internal/modules/cjs/helpers:93:18)
    at Object.<anonymous> (/tmp/.mount_arduinOp8oqp/resources/app/node_modules/keytar/lib/keytar.js:1:14)
    at Module._compile (node:internal/modules/cjs/loader:1110:14) {
  code: 'ERR_DLOPEN_FAILED'
}
```

Leaving this here for visibility until the AppImage gets fixed.