arduino/arduino-ide #361: automatically check for update to the IDE

jerabaul29, 2021-05-09:
For now, as far as I know, I have to 1) download the latest binaries 2) extract and move to the right place in order to get the IDE 2.0 in the latest version.

Would be great to have a built-in way to update the IDE, for example checking at each startup if we are using the latest version.

Attigliuzzo, 2021-05-13:
I think a similar option should be added to update automatically libraries and boards when an update is available.

Logicer16, 2021-06-05:
There's a check box in preferences though it's greyed out.

per1234, 2022-02-01:
I see we now have a duplicate request about this at https://github.com/arduino/arduino-ide/issues/772.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place. Even though this one came first, the other has more technical details, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-02-01 by per1234