arduino/arduino-ide #365: Can't connect to serial port of Raspberry Pi Pico

benevpi, 2021-05-10:
**Describe the bug**
I can program the Pico from Arduino IDE 2.0, but not connect to the serial port even when the serial port is enabled and producing output (that can be read from Arduino IDE 1.8

**To Reproduce**
Select Raspberry Pi Pico as the board. It will show as not connected, but this doesn't seem to stop us programming it.

Connect Pico, load example > Communication > ASCIITable

Load this to board (you may have to restart the Pico while holding down bootsel before trying to load the sketch)

One it's loaded, the board should have a serial output. If you go to Select Other Board and Ports  and check show all ports, the list is empty

If I now close the IDE 2.0 and open Arduino IDE 1.8, I can select the serial port and open the serial monitor, and the information is there correctly

**Expected behavior**
The serial monitor should open and display the details of the Ascii table

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.0 beta 5

benevpi, 2021-05-10:
It's started working. I don't know why it's started working, but it has.

per1234, 2021-11-03:
I'll go ahead and close this as either resolved or covered by https://github.com/arduino/arduino-ide/issues/573.

closed 2021-11-03 by per1234