arduino/arduino-ide #367: Please add the cursor column to the status line

RudolfAtHome, 2021-05-12:
All IDEs I use show the current line in the status line as well as the current cursor column. But the Arduino IDE only shows the line number. I am very grateful for addition of the column! It's more then enough room.
See examples:

[image]

Lorandil, 2021-05-13:
I'm missing that functionality too!
It would even save space in the code window because it would allow to disable the line numbers.

Daedelos, 2021-05-13:
And Textpad:
[image]

RudolfAtHome, 2021-05-16:
It is based on the Theia IDE framework... And the Theia IDE shows in its status line (right corner):
[image]

per1234, 2022-05-26:
For the sake of creating a cross-reference, I'll add a link here to the equivalent Arduino IDE 1.x issue: https://github.com/arduino/Arduino/issues/4403

kittaakos, 2023-05-10:
2.1.0 shows the line and column numbers.

https://github.com/arduino/arduino-ide/assets/1405703/75c44861-f0cb-495a-99c3-ed7c704ba7c7

closed 2023-05-10 by kittaakos