arduino/arduino-ide #369: Please allow overwrite mode in the editor

Daedelos, 2021-05-13:
In the IDE 1.x editor it is possible to edit in insert or overwrite mode - toggling between modes by using the Insert key.

Most editors allow the user to select insert or overwrite mode.

The IDE 2.x code editor mandates insert mode only.

barrydegraaff, 2023-10-28:
Switching back to version 1.8 for this one.

per1234, 2023-10-28:
Thanks for your request @Daedelos. This limitation originates from the Monaco editor used by Arduino IDE 2.x. The support of overtype mode must be added there, after which Arduino IDE would inherit the support as a matter of course. So I'll close this as off topic for this repository.

The work in the Monaco editor is already tracked at https://github.com/microsoft/vscode/issues/1012. If you browse through that thread you'll find a workaround mentioned. Although that discussion is about VS Code, the same workaround will work for Arduino IDE 2.x as well. We don't provide support for  the workaround here, but if try it and have questions or problems, you are welcome to post over on Arduino Forum. I'm sure we'll be able to help you out over there:

https://forum.arduino.cc/c/software/ide-2-x/93

closed 2023-10-28 by per1234