arduino/arduino-ide #373: Make 'bare minimum' sketch modifiable by users

dougp2, 2021-05-16:
https://forum.arduino.cc/t/modify-bare-minimum/862290

It's gone.  This was a nice thing in IDE 1.8.13.

JAY-M-ERICSSON, 2021-09-02:
The content of a new sketch is hardcoded

https://github.com/arduino/arduino-ide/blob/8c4e66f53628024a97581e668a8fde323192d195/arduino-ide-extension/src/node/sketches-service-impl.ts#L254-L263

that is not great… why not duplicate a default sketch (bareMinimum.ino but user definable would be ideal).
(This allows a new sketch to match personal preferences for what we want to see there. Anyone tired of deleting the two comments (not localized) and adding a Serial.begin for example ?

UKHeliBob, 2022-02-10:
What is the current status of this ?

I see that the issue is closed but has been assigned to someone.  At first sight that does not make sense

per1234, 2022-02-10:
Hi @UKHeliBob. The issue is not closed.

The "Closed" indicator you are seeing is for a duplicate here: https://github.com/arduino/arduino-ide/issues/733.

This is just the way GitHub works. When you link to an issue or PR from any comment on GitHub, it adds this "pingback" event to the referenced issue, with a label that shows the state of the referencing issue or PR. It does seem to cause some occasional misunderstandings from those who aren't super experienced with the GitHub framework, but it is actually a super useful feature that often leads me to valuable information.

UKHeliBob, 2022-08-01:
Can I have an update on this please ?

I am trying to use rc9 as my "daily driver" IDE but every time I start a new sketch the fact that the default new sketch cannot be modified annoys me.  For instance, I never need the comment lines in the sketch and have to delete them

Ideally the default sketch that is presented when starting a new sketch would be configurable but I would settle for it being BareMinimum as long as I can edit it to suit my working practices

closed 2022-10-26 by kittaakos