arduino/arduino-ide #375: Serial Monitor sends spurious data

per1234, 2021-05-16:
**Describe the bug**

Spurious data is sent by the Serial Monitor over the connected serial port.

**To Reproduce**

Steps to reproduce the behavior:

1. Paste the following simple serial echo sketch into the Arduino IDE:
    ```cpp
    void setup() {
      Serial.begin(115200);
    }

    void loop() {
      if(Serial.available()){
        Serial.println(Serial.read());
      }
    }
    ```
2. Upload to an Arduino Mega (I haven't been able to reproduce the bug with other boards I have)
3. Open Serial Monitor
4. Set the baud rate menu to 115200. The Serial Monitor's output field shows unexpected data is being received by the Arduino board and echoed back by the sketch, something like:
    ```
    240
    240
    240
    240
    240
    240
    ```
   ... and so on

**Expected behavior**

Data is only sent over the serial port by Serial Monitor when the user sends it via the input field.

**Desktop (please complete the following information):**

 - OS: Windows 10
 - Version: Version: 2.0.0-beta.6-nightly.20210515
Date: 2021-05-15T03:10:15.375Z
CLI Version: 0.18.2 alpha [7b5a22a4]

**Additional context**
On my Leonardo, the RX LED flashes at 1 Hz while the Serial Monitor is connected, indicating unexpected data is being received, but I don't get any output from the echo sketch.

I notice that the ~1 Hz frequency of the spurious data appears to match the frequency of the notification displayed repeatedly after the board is disconnected while Serial Monitor is open:

> Reconnecting Arduino Mega or Mega 2560 to COM18 in 1 seconds...

This makes me wonder if already open ports are constantly being reopened at 1 Hz the whole time Serial Monitor is running. Further evidence supporting this is that I get the RX blink on the Leonardo whenever I do a manual [`StreamingOpen` gRPC request](https://arduino.github.io/arduino-cli/latest/rpc/monitor/#cc.arduino.cli.monitor.v1.StreamingOpenRequest) to the port.

It appears the data is sent at 115200 baud.

Originally reported at: https://forum.arduino.cc/t/serial-monitor-sends-random-f0/850819

NazariiBoichuk, 2021-11-16:
I faced similar problem with Arduino Due on Arduino IDE 2.0.
The problem disappeared when I used usual Arudino IDE 1.8.16.
More about my issue here:
[https://forum.arduino.cc/t/serial-read-and-serial-available-are-working-not-as-expected-in-the-code/925823](https://forum.arduino.cc/t/serial-read-and-serial-available-are-working-not-as-expected-in-the-code/925823)

hemmikall, 2021-12-16:
Still seems to be an issue.

Using Arduino UNO on COM3, the serial monitor sends out two bytes of 0xF0 (hex value of each byte) every second.
As soon as I close the serial monitor, the transmission stops.

BlueBles, 2022-03-03:
Still a problem. 
Baud rate 115200
Arduino Mega on IDE 2.0 recaive some weird random data on serial. 
On old IDE, no problem, all text from serial is read good.

PaulStoffregen, 2022-03-03:
Maybe switching to new API (#769) will help?  Or at least should be done first, so testing and troubleshooting is done with new pluggable serial monitor?

sterretjeToo, 2022-07-30:
Some additional information that might help; with the given sketch
1) SparkFun RedBoard (FTDI) and clone Nano (CH340) do not show activity on the RX led and no data is displayed in the serial monitor.
2) The RX LED flashes on an Arduino Leonardo / Micro as well as on a SparkFun ProMicro; nothing is displayed in serial monitor. Specifically on the ProMicro this can be disturbing as the RX LED is used is often used as a heartbeat to detect if a sketch might be hanging or not.

nhatnamt, 2022-08-14:
I faced a similar problem with Arduino Due at 9600 baud. The same code works fine on 1.8

MikeGWem, 2022-09-06:
Still an issue with RC9.3
Simple program attached. Works as expected in V1 IDE but collects spurious data when the baud rate is set to 115200 in V2 IDE
Workaround is to set a baud rate of 9600. [Note when testing, "No Line Ending" should be set on Serial Monitor. Tested board - Uno
[HelloWorld2.zip](https://github.com/arduino/arduino-ide/files/9499702/HelloWorld2.zip)

palazzol, 2022-11-08:
Hello,

I am currently having this issue with an Arduino Uno R3.  It looks like the IDE is sending redundant changes in configuration data every second, and even more when you select the pulldowns on the serial monitor. (Not sure exactly what, but it could be like setting the baud rate, stop bits, etc.) For some boards, this has no ill effects.  But for the ones with an ATMEGAU2 at least, this can cause the firmware to momentarily reset the UART, putting spurious pulse out on the Serial receive to the 328P, in my case.  I can clearly see the pulses on an oscilloscope.  Note, the RXD light driven by the ATMEGAU2 is not flashing - it's not "received data" being sent, it's something else causing spurious pulses.  This is why they oddly seem to show up as mostly 0xf0 at 115200 baud - It's just the luck of the timing.  Strangely, the only time it seems to stop is when you select 57600, which I can see is a special case in the ATMEGAU2 firmware.  

Anyway, my Javascript is rusty but I think this could be fixed by finding and eliminating the redundant config information being sent by the Serial Monitor.  Alternatively, I bet you could also update the ATMEGAU2 firmware to check the incoming state changes to see if they are needed, and not reset the UART unless required.  

I want to learn how to build the new IDE, and then I can try to work on this some more.  I'd also be happy to test any updates you may be evaluating.

palazzol, 2022-11-08:
By the way, the pulses seem to come in pairs.  Both of which usually, but not always, decode as an 0f0 at 115200 baud.  It may be the UART being turned off, and then turned back on.  My guess is that this code is being triggered on the ATMEGA16U2:

[https://github.com/arduino/ArduinoCore-avr/blob/master/firmwares/atmegaxxu2/arduino-usbserial/Arduino-usbserial.c#L204](https://github.com/arduino/ArduinoCore-avr/blob/master/firmwares/atmegaxxu2/arduino-usbserial/Arduino-usbserial.c#L204)

MikeGWem, 2022-11-08:
This is entirely a problem associated with the IDEv2 implementation of the Serial Monitor. Version 1 IDE - all is fine. I even built my own implementation of a Serial Monitor in C#/Windows Forms and had none of the issues we are seeing here. No issue when using putty either.

palazzol, 2022-11-12:
Well, I've made some more progress.  The Arduino-IDE program uses a component named arduino-cli.exe, which in turn uses another component named serial-monitor.exe to talk to the ATMEGA16U2 on the Arduino Uno.  I have run serial-monitor.exe by itself, and after it executes the OPEN command, the 1Hz glitch pairs start appearing on the Uno's RXD line.  There does not appear to be any command being executed every second, however.  I am going to dig into the firmware on the ATMEGA16U2 next, and try to figure out what is really going on there.  The IDE isn't completely off the hook, because it is still sending too much data to the board, especially when using the Line Endings pulldown - which shouldn't even affect the serial port.  But the root cause of the periodic corruption has been narrowed down to an interaction between serial-monitor.exe and the ATMEGA16U2 firmware.

palazzol, 2022-11-15:
1) The root cause of the periodic corruption is inside the go serial library used by serial-monitor.exe.  I submitted a pull request above which I believe takes care of it.  I'm not sure how long it will take for that to be accepted and then used by arduino serial-monitor though.  The Arduino project may want to fork it if the fix is not adopted quickly.  2) The extra corruption caused by the flurry of comm updates when various pulldowns are used has been reduced with another pull request.  This limits the flurry to only the baud rate pulldown.  **I still couldn't figure out how to change only the baud rate, without changing all the other serial parameters, but I hope someone who reads that pull request will be able to help there.**  I still want to do one more fix to the ATMEGA16U2 firmware, to mask any glitches due to legitimate reconfiguration.  It looks like the original LUFA USB-serial example had that feature, but it didn't make it into the 16U2 code for some reason.

MikeGWem, 2022-11-15:
Well done Frank Palazzolo. A solution in sight.

kittaakos, 2022-11-29:
I am reopening as the #1703 only partially fixes the issue: https://github.com/arduino/arduino-ide/pull/1703#pullrequestreview-1197517996

Per, please close it if you think this is resolved. Thank you!

per1234, 2022-12-01:
Fixed by https://github.com/bugst/go-serial/pull/147 + https://github.com/arduino/serial-monitor/pull/31.

These fixes are in the [**serial-monitor** 0.11.0 release](https://github.com/arduino/serial-monitor/releases/tag/v0.11.0), which will be automatically installed the next time the user starts Arduino IDE 2.x.

commit de3794a3 (2022-11-23): fix: update monitor settings only if it's changed

Closes #375
commit 0a675913 (2022-11-28): fix: update monitor settings only if it's changed

Closes #375
commit 2d5df272 (2022-11-28): fix: update monitor settings only if it's changed

Closes #375
commit 3403e6fe (2022-11-28): fix: update monitor settings only if it's changed

Closes #375
closed 2022-12-01 by per1234