arduino/arduino-ide #379: error "exec: "cmd": executable file not found in %PATH%"

KevWal, 2021-05-16:
**Describe the bug**
Clicking Verify works fine until the last command, which it tries to run with 'cmd /C', which fails with an error "exec: "cmd": executable file not found in %PATH%"

```
"C:\\Users\\WALTONK\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\WALTONK\\AppData\\Local\\Temp\\arduino-sketch-CD01B93046678F34858CBF37069BDA38/Blink.ino.elf" "C:\\Users\\WALTONK\\AppData\\Local\\Temp\\arduino-sketch-CD01B93046678F34858CBF37069BDA38/Blink.ino.hex"

cmd /C "C:\\Users\\WALTONK\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\WALTONK\\AppData\\Local\\Temp\\arduino-sketch-CD01B93046678F34858CBF37069BDA38/Blink.ino.elf" > "C:\\Users\\WALTONK\\AppData\\Local\\Temp\\arduino-sketch-CD01B93046678F34858CBF37069BDA38/Blink.ino_atmega1284p_11059200L.lst"
Compilation error: Error: 2 UNKNOWN: exec: "cmd": executable file not found in %PATH%
```

**To Reproduce**
Steps to reproduce the behaviour:

Load the Blink example, click Verify, and I get the error.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: Arduino IDE 2.0.0-beta.6

**Additional context**

KevWal, 2021-05-16:
cmd.exe lives in C:\Windows\System32 which was not in my path, but Arduino assumes it will be in the path.

Added C:\Windows\System32\ to my path, restarted the IDE, and now it works.

closed 2021-05-16 by per1234

KevWal, 2021-05-16:
Should the software be relying on Windows path to.include the path to cmd.exe, if my system was missing it, might others not also be?

Is there a reason the last command needs to be run within cmd.exe, when all the others do not?

Thanks
Kevin

per1234, 2021-05-16:
>Should the software be relying on Windows path to.include the path to cmd.exe, if my system was missing it, might others not also be?

I think it's probably reasonable for the platform developer to assume Windows users will have this in their path.

>Is there a reason the last command needs to be run within cmd.exe, when all the others do not?

I guess it's to support the redirect (`>`)

Anyway, none of this has anything to do with the Arduino IDE. This is about the 3rd party MightyCore platform you're using, and so completely off topic for discussion in this repository. If you have a suggestion for an improvement to MightyCore, feel free to submit a pull request to that repository.

KevWal, 2021-05-17:
Thanks for the information, I didn't realise this was Mighycore specific.