arduino/arduino-ide #380: Monaco editor: some keywords (type, CONSTANT) are not properly coloured

ubidefeo, 2021-05-18:
**Describe the bug**
some keywords/types/CONSTANTS do not get proper colouring in the Monaco editor

**To Reproduce**
Steps to reproduce the behavior:
- Create a Sketch
- replace its content with the following
```C++
byte myNumber = 0;
void setup(){
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(3, INPUT);
}
void loop(){
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
}
```

**Expected behavior**
`byte`, `OUTPUT`, `INPUT`, `LOW`, `HIGH` should be coloured as type and CONSTANTS

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows/Mac
 - Version: 2.0.0 Beta 7

mklemarczyk, 2022-04-06:
Any progress on this one? It has been almost a year.

eMUQI, 2022-10-08:
Is this a feature or a bug? The Arduino 2.0.0 stable version now released on the Arduino official website still has this problem. Any progress on this?

60-hz, 2023-08-01:
It would be nice to fix this issue, I am using it for teaching and I miss this syntax color :)