arduino/arduino-ide #381: V2 beta 4 and up, does not startup and is only showing the splashscreen

nielsnl68, 2021-05-18:
**Describe the bug**
After starting i get the splashscreen and the window with a running wait icon.
noting happens after this.

**To Reproduce**
Steps to reproduce the behavior:
1. Just start the the EDI.
2. no errors found in the console. 
3. when i try to open en skatch it crashes with a message behind the splashscreen.

**Expected behavior**
That it atliest removed the splashscreen. and gives an overview of the workwindow,

**Desktop (please complete the following information):**
 - OS: latest windows 10. 
 - version from nodejs installed is v14.16.1

**Additional context**
I like to help out on finding what is happening. when someone can politely ask the right questions i will do what i can to updated this bug report

NikilXYZ, 2021-05-18:
Have experienced a similar issue on Surface Go laptops and some older Acer laptop, Windows 10 (student computers).

Seems to hang on that splash screen

NikilXYZ, 2021-05-18:
For the record, the IDE opened yesterday. Not sure what changed

nielsnl68, 2021-05-18:
Did you do an upgrade?
Which version did you use when it worked fine?

kittaakos, 2021-05-20:
> 3\. when i try to open en skatch it crashes with a message behind the splashscreen.

Could you please start the IDE from a `CMD.exe` (instead of clicking on the application icon) and share the terminal output with us after the app crash? Thank you!

nielsnl68, 2021-05-20:
Here are the votes of the Netherlands: (Hmmmm... i mean hear is the output of the cmd line:

```
Starting backend process. PID: 17596
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root WARN please install @theia/electron@1.12.0-next.1a1d7afc as a runtime dependency
root INFO Configuration directory URI: 'file:///c%3A/Users/HP/.theia'
root INFO Theia app listening on http://localhost:51254.
root WARN The local plugin referenced by local-dir:C:\Users\HP\.arduinoProIDE\plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/HP/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/HP/.arduinoIDE/extensions does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.1-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\plugins\cortex-debug\extension\dist\extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.1-beta.1" from "C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-cpp\extension"
root INFO Deploying backend plugin "json@1.46.1" from "C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-json\extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
root INFO Deploy plugins list took: 502.8 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 1895.5 ms
daemon INFO Starting daemon from C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
root WARN Backend ArduinoDaemonImpl.onStart is slow, took: 139.7 ms
daemon INFO Using config file: c:\Users\HP\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.18.3
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Daemon is running.
root INFO Git askpass helper is listening on http://127.0.0.1:51263.
root WARN Frontend CommonFrontendContribution.configure is slow, took: 1286.9 ms
root WARN Frontend commands.onStart is slow, took: 183.2 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/safer-buffer
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/uuid
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @types/p-queue
root WARN Failed to resolve module: @theia/preferences
root WARN Failed to resolve module: @theia/electron
root WARN Failed to resolve module: @theia/keymaps
root WARN Failed to resolve module: @theia/messages
root WARN Failed to resolve module: @types/diff
root WARN Failed to resolve module: @types/dompurify
root WARN Failed to resolve module: @types/mime
root WARN Failed to resolve module: @types/request
root WARN Failed to resolve module: @types/dateformat
root WARN Failed to resolve module: @types/deepmerge
root WARN Failed to resolve module: @types/glob
[... 127 lines cut ...]
daemon INFO Loaded tool [tool: Version: 1.22.0-97-gc752ad5-5.2.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
daemon INFO Loading package esp8266 from: C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\hardware
daemon INFO Loaded platform [platform: {ID:"esp8266:esp8266",Installed:"3.0.0",Latest:"3.0.0",Name:"esp8266"}]
daemon INFO Checking existence of 'tools' path: C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools
[... cut ...]
    at Object.onReceiveStatus (C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\node_modules\@grpc\grpc-js\build\src\client.js:387:49)
    at Object.onReceiveStatus (C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\node_modules\@grpc\grpc-js\build\src\client-interceptors.js:299:181)
    at C:\Users\HP\AppData\Local\Programs\arduino-ide\resources\app\node_modules\@grpc\grpc-js\build\src\call-stream.js:145:78
    at processTicksAndRejections (internal/process/task_queues.js:76:11)
daemon INFO Daemon exited with exit code: 1.
```

let of warning as i saw looking around. I hope it helps.

kittaakos, 2021-05-20:
Thank you! This is the problematic part from the logs:
```
daemon ERROR Error during sketch processing: Filesystem bottom is too deep (directory recursion or filesystem really deep): c:\Users\HP\Documents\Arduino\sketch_jan31a\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader\rfid-reader
root ERROR Uncaught Exception:
```

The IDE cannot start because there is a folder-structure recursion under: `c:\Users\HP\Documents\Arduino\sketch_jan31a\rfid-reader\rfid-reader`. If you know how to reproduce it, or if you remember what you did with the IDE, please share the steps so that someone can fix it.

> always been on beta 2.0 v4

😕 Unfortunate. A recursion-related [issue](https://github.com/arduino/arduino-ide/pull/211) was fixed as part of [`2.0.0-beta.4` ](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-beta.4) the release.

I had the same issue previously on Windows, and there is a workaround. I had to wipe the corrupted folder structure manually. Unfortunately, I am not aware of any kind of solution other than deleting `c:\Users\HP\Documents\Arduino\sketch_jan31a\rfid-reader\rfid-reader` from your file system.

For me, it was tricky. Deleting the file recursion took forever on Windows. I tried with Explorer, `CMD.EXE`, and Total Commander. Finally, I ended up using Git Bash for deletion. If you use Git Bash, you have to do the followings to delete the folder recursively:

**Please make sure to make a backup before you perform some unrecoverable deletion**

```sh
rm -rf ~/Documents/Arduino/sketch_jan31a/rfid-reader/rfid-reader
```

@per1234, do you happen to know any other less invasive solution than `rm -rf`? Thank you!

nielsnl68, 2021-05-21:
> I had the same issue previously on Windows, and there is a workaround. I had to wipe the corrupted folder structure manually. Unfortunately, I am not aware of any kind of solution other than [...]
Owww, i should have read the whole message, then i would not have taken so much time to try the same options. I managed to move the folder away from the Arduino skatch folder, after that arduino-ide started up again.

> **Please make sure to make a backup before you perform some unrecoverable deletion**
> 
> ```shell
> rm -rf ~/Documents/Arduino/sketch_jan31a/rfid-reader/rfid-reader
> ```
> 
> @per1234, do you happen to know any other less invasive solution than `rm -rf`? Thank you!

Thanks for your help, sadly the 'rm -rf' solution was not working for me. I'm getting a 'File name too long' message.  So for me de damage is more serious, then yours.
I think

nielsnl68, 2021-05-21:
@dev's could you share the code that created this issue. Maybe it can be used to resolve the issue as well. when it can create so many sub-folders maybe it can delete them as well.

kittaakos, 2021-05-21:
> I'm getting a 'File name too long' message.

Please share the entire error/output so that we can figure out what's wrong. Thanks!

nielsnl68, 2021-05-21:
noting much only a lot of `\rfid-reader` after i called `rm -rf <recursingfolder>` in the `git bash` as you suggested above. Btw. The error message is from `rm -rf`
So the number of recursion folders was just to much for rm.

per1234, 2021-05-29:
Hi all. I apologize for being so slow to respond. I had this same problem and also had trouble removing the deep folder structure.

I had resigned myself to having the folder as a permanent memento of my beta testing work then when I was providing support to someone on the forum who had the same problem it was mentioned that `rm -rf` with Git Bash had worked for them to remove it. Even though I had already tried it without success, I gave it another try and was pleasantly surprised to find that it worked. My hypothesis was that some process attempting to deal with the difficult folder structure had gone zombie and locked it. At some point after my initial attempts, I restarted my computer, which killed the process and unlocked the folder, allowing it to be deleted. 

I just did some experiments and, after a few false starts when I was able to delete the folders normally without any problems, was finally able to reproduce the problem of the impossible to delete folder structure. I restarted my computer and sure enough I was then able to delete it.

So please restart your computer and then try another attempt at `rm -rf` @nielsnl68. After that, please let us know what the results are.

Ho-Ro, 2021-06-25:
This is the same behaviour as already mentioned in #351 (big directory structures with circular links exist in sketchbook directory).

nielsnl68, 2021-06-25:
True, at the time of posting, i did not know what the issue was. Later someone pointed me to that same issue.

Still waiting for a good solution from the Arduino-edi team, to fix this.

Ho-Ro, 2021-06-25:
Yes, it's nasty since I organize my documents and directories heavily with (criss-cross) softlinks.

per1234, 2022-09-22:
Since we have not received any additional reports from users who were affected by https://github.com/arduino/arduino-ide/pull/211 in over a year, I think this issue has served its purpose of documenting remediation measures. So I will close this.

We will track the need for the IDE to gracefully handle problematic sketchbook folder structures intentionally produced by the user in https://github.com/arduino/arduino-ide/issues/351.

closed 2022-09-22 by per1234