arduino/arduino-ide #385: The startup takes forever.

Liferme, 2021-05-19:
**Describe the bug**
When starting up the APP, it frozes with the splash screen and a running icon beind, it always takes 10mins or more to get fully loaded.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Applications
2. Click on 'Arduino IDE'
3. See error

**Expected behavior**
Only few seconds to get load up.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Big Sur (11.3)
 - Version: Arduino IDE 2.0.0 beta7

kittaakos, 2021-05-20:
What's the error if you start the IDE from the terminal? To start the IDE from a terminal, do the followings:
 - Open _Terminal.app_ and
 - Run `/Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE`.

Liferme, 2021-05-23:
> What's the error if you start the IDE from the terminal? To start the IDE from a terminal, do the followings:
> 
> * Open _Terminal.app_ and
> * Run `/Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE`.

The error still exists, it still takes moren than 5mins to load up.

kittaakos, 2021-05-23:
What do you see in the terminal? Can you share the output?

esp32croft, 2021-05-25:
Hi kittaakos,
I'm also using a Mac running Big Sur. However I don't get a 5-10 minute delay. More like abut 1 minute (subjective, could well be less).
I've attached my terminal log FYI, I don't know if you have many Mac users in your development/test teams.
[StartupLog.txt](https://github.com/arduino/arduino-ide/files/6538187/StartupLog.txt)

I've also included a log from my 10 year old MacBook running MacOs Sierra which also does not take 10 minutes.
[MacBook startupLog.txt](https://github.com/arduino/arduino-ide/files/6538308/MacBook.startupLog.txt)

Both show comments about certain processes being slow, and other warnings.
Hope that helps.

Ho-Ro, 2021-06-25:
maybe related to #351 / #381?

Wolfy7, 2021-08-10:
Hello all.

I also have the same issue. 
On Windows Version 21H1 (Build 19043.110) the startup dosent finish, the spalsh screen stays forever. Some entries from the menu bar are working for example "Help -> About Arduino IDE", but not all like "File -> New" dosent work.

[image]

[StartupLog.txt](https://github.com/arduino/arduino-ide/files/6960364/StartupLog.txt)

Version: 2.0.0-beta.10
Date: 2021-08-05T08:23:32.577Z
CLI Version: 0.18.3 alpha [d710b642]

ckuethe, 2021-09-17:
Looks like it spends a lot of time checking for library and board updates... and retrying in case of error. Seems like these checks need to be more asychnronous.

"root ERROR Error while updating index in attempt 9."

Full log at https://gist.github.com/ckuethe/e3a4accf142e7668eb736a808c79e11c

pierreverbakel, 2021-10-05:
These library checks can be done, and the results of those findings can be stored in json-files, a light database and loaded next time from there if after a quick check it is clear that there has been a change in the libraries. Personally I have been loosing the time of a good holiday waiting for the Arduino IDE coming up in previous version, staring at the screen as if tomorrow does not exist, wondering who in earth makes this to be so slow and is able not to itch its brain to rethink the necessity of all the initialisation checks, and if maybe the outcome of these checks is 99% the same as in other starting up of the IDE. Ideas like caching of results come to mind, databases as stated, or simply an option in the menu for the user to update the Library-Database. It is all a bit like a car starting by itself when you walk to the grocery-store, since maybe you want to use the car for that and then the car needs to be up and running. This costs all wasteful gasoline, or wasteful CPU, HD-time and what not. 
Long story short: Needs heavely improved. Also in 1.8.x

gwachob, 2021-11-26:
I'm seeing the same problems. It appears that the IDE is looking to bind to port 50051 and failing repeatedly on startup (when launched from command line with `/Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE`):

> daemon INFO Restarting daemon in 5 seconds...
daemon INFO Failed to listen on TCP port: 50051. Address already in use.

Doing some sleuthing, it looks like at least the multipass (Ubuntu virtual environment from Canonical) daemon starts uses launchd to bind to this port: 

> $ find ~/Library/LaunchAgents /Library/LaunchAgents /Library/LaunchDaemons /System/Library/LaunchAgents /System/Library/LaunchDaemons -name "*.plist" -exec grep -H 50051 "{}" \; 2>/dev/null
  /Library/LaunchDaemons/com.canonical.multipassd.plist:            <string>50051</string>

Unfortunately, just stopping the multipass daemon (using the Multipass's menubar widget) doesn't make the issue go away (since this is auto-launchd by launchd in the Mac and launchd appears to bind on port 50051). 

I don't know what the solution here, and I'm not even 100% this is causing the launch times to be so slow, but hopefully this helps a bit.

per1234, 2021-11-27:
Hi @gwachob. The situation with the gRPC port collision is discussed here:
https://github.com/arduino/arduino-ide/issues/201

>I don't know what the solution here, 

The real solutions are discussed at https://github.com/arduino/arduino-ide/issues/201 and https://github.com/arduino/arduino-cli/issues/1260, but I'll provide instructions you can follow to resolve the port conflict:

1. Open this file in a text editor:
   ```
   ~/.arduinoIDE/arduino-cli.yaml
   ```
1. Change the port number in the `daemon.port` key to some port that is not already in use (e.g., `"50052"`).
1. Save the file.

Now try starting the Arduino IDE again as before. It should now start up and work as expected.

Farhankhosravi, 2022-01-03:
Hi
it opens very slow for me too
i'm using rc3 version, on windows 11
this is the log file while app is opening:

`Starting backend process. PID: 6892
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/Farhan/.theia'
root ERROR (node:6892) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
root INFO Theia app listening on http://localhost:3229.
root WARN The local plugin referenced by local-dir:/c%3A/Users/Farhan/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/Farhan/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\Farhan\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "D:\Program Files\Arduino IDE\resources\app\plugins\cortex-debug\extension\dist\extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.2-beta.1" from "D:\Program Files\Arduino IDE\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "D:\Program Files\Arduino IDE\resources\app\plugins\vscode-builtin-cpp\extension"
root INFO Deploying backend plugin "json@1.46.1" from "D:\Program Files\Arduino IDE\resources\app\plugins\vscode-builtin-json\extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "D:\Program Files\Arduino IDE\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
root INFO Deploy plugins list took: 856.1 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 2773.2 ms
daemon INFO Starting daemon from D:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
daemon INFO Using config file: c:\Users\Farhan\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.20.2
daemon INFO Executing `arduino-cli daemon`
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO {"IP":"127.0.0.1","Port":"3230"}
daemon INFO Daemon is running.
root INFO Git askpass helper is listening on http://127.0.0.1:3241.
daemon INFO Adding libraries dir [dir: c:\Users\Farhan\Documents\Arduino\libraries, location: user]
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root ERROR Error: Git installation not found.
    at D:\Program Files\Arduino IDE\resources\app\node_modules\find-git-exec\lib\find-git-exec.js:69:69
    at async DefaultGitInit.init (D:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\lib\node\theia\git\git-init.js:22:49)
root WARN Frontend CommonFrontendContribution.configure is slow, took: 434.0 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/dompurify
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/safer-buffer
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/multer
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/uuid
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @types/p-queue
root WARN Failed to resolve module: @theia/preferences
root WARN Failed to resolve module: @theia/editor-preview
[... 134 lines cut ...]
daemon INFO Resolving include Adafruit_Sensor.h for arch avr
daemon INFO   found better lib [lib: Adafruit_Unified_Sensor, prio: 3EB]
daemon INFO Resolving include TM1637TinyDisplay.h for arch avr
daemon INFO   found better lib [lib: TM1637TinyDisplay, prio: 643]
[... cut ...]
daemon INFO Resolving include SPI.h for arch avr
daemon INFO   found better lib [lib: SPI, prio: 64C]
daemon INFO Resolving include Adafruit_BMP280.h for arch avr
daemon INFO   found better lib [lib: Adafruit_BMP280_Library, prio: 517]
daemon INFO Resolving include Adafruit_Sensor.h for arch avr
daemon INFO   found better lib [lib: Adafruit_Unified_Sensor, prio: 3EB]
daemon INFO Resolving include TM1637TinyDisplay.h for arch avr
daemon INFO   found better lib [lib: TM1637TinyDisplay, prio: 643]
`

Farhankhosravi, 2022-01-08:
> Hi it opens very slow for me too i'm using rc3 version, on windows 11 this is the log file while app is opening:
>
> ```
> Starting backend process. PID: 6892
> Using browser-only version of superagent in [...]

Someone help me please?!

per1234, 2022-01-08:
>Someone help me please?!

@Farhankhosravi this is not an appropriate place to ask for help. This is the place to contribute to the development of Arduino IDE 2.x. That can include submitting formal bug reports, or to provide additional information that will help with the investigation of the existing reports.

We do not provide support here. That is done on the Arduino Forum, which has a dedicated category just for Arduino IDE 2.x:

https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93

raffeale, 2022-02-14:
the Arduino 2.0 ide startup needing 2 minutes, I start it form terminal, my Laptop is MacBook Pro (Apple Silicon)
I found it spend a lot time to update index ,following is the log

daemon INFO Checking signature [error: opening signature file: open /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/710196979.sig: no such file or directory, index: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/710196979, signaturefile: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/710196979.sig]
daemon INFO URL: 
daemon INFO Updating index [url: ://]
root ERROR Error while updating index in attempt 1.
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root INFO Updating index: package_index.json completed.
root INFO Updating index: package_index.json.sig completed.
daemon INFO Checking signature [error: opening signature file: open /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/883332696.sig: no such file or directory, index: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/883332696, signaturefile: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/883332696.sig]
daemon INFO URL: 
daemon INFO Updating index [url: ://]
root ERROR Error while updating index in attempt 2.
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root INFO Updating index: package_index.json completed.
root INFO Updating index: package_index.json.sig completed.
daemon INFO Checking signature [error: opening signature file: open /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/914061345.sig: no such file or directory, index: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/914061345, signaturefile: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/914061345.sig]
daemon INFO URL: 
daemon INFO Updating index [url: ://]
root ERROR Error while updating index in attempt 3.
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root INFO Updating index: package_index.json completed.
root INFO Updating index: package_index.json.sig completed.
daemon INFO Checking signature [error: opening signature file: open /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/636154398.sig: no such file or directory, index: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/636154398, signaturefile: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/636154398.sig]
daemon INFO URL: 
daemon INFO Updating index [url: ://]
root ERROR Error while updating index in attempt 4.
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root INFO Updating index: package_index.json completed.
root INFO Updating index: package_index.json.sig completed.
daemon INFO Checking signature [error: opening signature file: open /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/726413407.sig: no such file or directory, index: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/726413407, signaturefile: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/726413407.sig]
daemon INFO URL: 
daemon INFO Updating index [url: ://]
root ERROR Error while updating index in attempt 5.
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root INFO Updating index: package_index.json completed.
root INFO Updating index: package_index.json.sig completed.
daemon INFO Checking signature [error: opening signature file: open /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/763113012.sig: no such file or directory, index: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/763113012, signaturefile: /var/folders/q_/jnr0_wxs5476p4br7_851j4r0000gn/T/763113012.sig]

ckuethe, 2022-02-26:
@rsora @Liferme @per1234 @cmaglie can this be prioritized? It's really painful having to wait 30 seconds for the IDE to check for library updates every time I start. I wouldn't mind waiting for updates to load when I specifically poll for it, but not when I just want to start editing code...

Farhankhosravi, 2022-03-03:
**SOMEONE HELP US!!...**
It's impossible to use an IDE with such SLOW opening...
i gave up using ide v2.0 and preferred to use old ide.. just because of this slowness

please fix this!!! it's very more important than your weekly cheap updates...... uh

hugo8873, 2022-05-11:
Same here, I'm on a MacBook Pro with M1-pro apple silicon, it takes more than 15 mins to start. 
After I removed the json url, a ESP32 json from espresso and a 3rd party STM32 json, it took 1 min.
I assume the problem is related with the json url.
Hope this helpful.

kittaakos, 2022-08-26:
Please try with the nightly. This issue was a duplicate of #876 and was resolved with #1018.

Please comment on this; if the issue is still present, I am happy to reopen or open a follow-up. Thanks!

closed 2022-08-26 by kittaakos

AlexW00, 2022-09-14:
> Please try with the nightly. This issue was a duplicate of #876 and was resolved with #1018.
> 
> Please comment on this; if the issue is still present, I am happy to reopen or open a follow-up. Thanks!

I am using the latest nightly version on my MacBook 2020 (intel base model i5), and startup takes about a minute.

kittaakos, 2022-09-14:
Thank you for reporting it @AlexW00. Maybe it's #1428. Can you please share the logs with us?

The easiest way to get the logs is by starting IDE2 from a terminal. If you open a `Terminal` on macOS and start the IDE2 with the following command, you will see the logs in a new `Terminal` window.

```sh
open /Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE
```

In my macOS, IDE2 is installed to `/Applications`. The command 👆 might be different based on your IDE2 install location.

The log starts with something like this:
```
/Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE ; exit;
a.kitta@Akoss-MacBook-Pro ~ % /Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE ; exit;
Starting backend process. PID: 89429
```

AlexW00, 2022-09-15:
Sorry, my bad. Looks like I did not install the new nightly version correctly, and I still opened the old build. The new nightly version starts in seconds. Thank you for the improvement!