arduino/arduino-ide #387: Memcpy function not recognized

Markz878, 2021-05-20:
When using the memcpy function, IDE complains that "Use of undeclared identifier 'memcpy'clang(undeclared_var_use)".

Markz878, 2021-05-20:
I noticed that in another function the memcpy function intellisense works correctly. Same thing with "Serial" class intellisense, it's only broken in one of the functions.. I'm using M-Duino family board.

per1234, 2021-12-13:
Thanks for your report @Markz878. There have been some recent improvements to the [Arduino Language Server](https://github.com/arduino/arduino-language-server) that provides this capability in the IDE. These improvements are available from the latest release of the Arduino IDE.

Please try again with [the latest build](https://github.com/arduino/arduino-ide#nightly-builds) of the Arduino IDE and let us know whether or not the bug is fixed for you.

per1234, 2022-03-11:
Closing due to lack of response.

@Markz878 if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-03-11 by per1234