arduino/arduino-ide #389: Posibility to open .ino files via explorer

fm-sys, 2021-05-24:
Currently, if you select the Arduino IDE 2.x as target app in the Windows explorer, nothing happens. Arduino IDE 2.x just starts without loading the file. Would be pretty cool if files can be opened via explorer instead of only inside the software.

kittaakos, 2021-05-25:
Same as: https://github.com/arduino/arduino-ide/issues/72
Related: https://github.com/arduino/arduino-ide/issues/370

per1234, 2021-05-25:
Thanks Akos!

Since @fm-sys was kind enough to add a note to https://github.com/arduino/arduino-ide/issues/72 about it not being macOS-specific, I'll go ahead and close this as a duplicate. It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place.

closed 2021-05-25 by per1234