arduino/arduino-ide #390: Ctrl + Shift + I is mapped twice

Stig124, 2021-05-25:
**Describe the bug**

`Ctrl` + `Shift` + `I` should be mapped as `Format Document` but it is also mapped to `Toggle Library Manager View`

**To Reproduce**

1. Do the combination

**Expected behavior**

It should format the file, not open the `Library Manager` view or the opposite (which it does but it must be made clearer)

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: OpenSUSE Tumbleweed 20210522
 - Version: 2.0.0-beta.7

kittaakos, 2021-05-25:
> `Ctrl` + `Shift` + `I` should be mapped as `Format Document`

Out of curiosity, why do you suggest it should map like that? `Ctrl/Cmd` + `T` is the format document. `CtrlCmd` + `Shift` + `I` was always Toggle Library View. I think we should rebind the default format document command, that comes from Theia, to `CtrlCmd` + `T` so that we provide the same UX as the Java-based Arduino IDE.

Stig124, 2021-05-25:
Hello,
The fun thing is that the arduino-native format command is alread mapped to `CtrlCmd` + `T` (command name `arduino-auto-format`), but the Theia-native takes preference when not using the `Tools` menu but using the `F1` as it doesn't appear as a command here.

Frankly, both would be good to me, but I would prefer the Theia-Native shortcut as I use more VSCode than the Java IDE

ubidefeo, 2022-07-29:
The __format__ command in IDE 2.0 will remain CTRL/CMD + T and Theia's "format document" command (accessible from the command palette) is not supported

closed 2022-07-29 by ubidefeo