arduino/arduino-ide #394: Starting IDE creates "undefined" folder in home directory

JayPi4c, 2021-06-01:
**Describe the bug**
It is more an oddity than a bug.
Whenever I start the IDE the program creates a folder called "undefined" in my home directory.  This folder contains only one folder called "Arduino IDE" which contains logs of the IDE. I think it's good that the logs are saved in an easy to find location. But all the time I'm looking into my home-folder the "undefined" folder kind of confuses me. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start the IDE
2. go to home folder
3. be happy with "undefined" folder

**Expected behavior**
Just giving it a more precise name would make me happy.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu 20.04.2 LTS
 - Version: 2.0.0-beta.7

**Additional context**
As already mentioned in #110 it is not perfectly documented how to install the IDE correctly, so i guess it could also be an error of me in the installation process.  In my point of view I went a straight forward way, downloading the newest IDE, unzipping the contents and copying the files to /opt/
To have it listed in the applications menu of Ubuntu I created the file `Arduino 2.0 (beta).desktop` in `~/.local/share/applications` with the following content:
``` 
[Desktop Entry]
Type=Application
# desktop entry spec version
Version=6.4.2
Name=Arduino 2.0 (beta)
Comment=Arduino IDE 2.0 beta
Exec=/opt/arduino-ide_2.0.0-beta.07/arduino-ide
Icon=/opt/arduino-1.8.13/lib/arduino.png
MimeType=image/svg+xml;
Terminal=false
Categories=Development
X-Desktop-File-Install-Version=0.24
```
It is mostly a copy of another .desktop file in the directory.

Now starting the IDE results in generating the undefined-folder. Everything else works perfectly.

ralphtheninja, 2021-06-17:
FWIW, I noticed this bug as well and did some digging. I think this is actually a bug inside Theia. But there's just too much typescript interface over engineered junk in the way and I could not find where a simple directory is created.

ggl, 2021-11-13:
It's here:

arduino/resources/app/node_modules/node-log-rotate/lib/utils/find-log-file-name.js

The default_1 function returns this log file name, with an obviously undefined path. It annoyed me to a great extent so I commented the return line and it doesn't do it any more.

LordRafa, 2022-02-17:
The error is related to an issue on the module node-log-rotate.

I have opened a PR with a fix, you can find more info here https://github.com/lemon-sour/node-log-rotate/issues/23

iwmq, 2022-03-09:
The `undefined` directory is still created for Arduino IDE 2 RC4. I found a workaround for this.

First, go to `Arduino IDE 2 installation directory`, open the file
`resources/app/node_modules/node-log-rotate/lib/utils/find-log-path.js`.

Then, find the following lines:

```
case 'linux':
	dir = prepareDir(process.env['XDG_CONFIG_HOME'] + '', appName)
		.or(process.env['HOME'] + '/.config', appName)
		.or(process.env['XDG_DATA_HOME'] + '', appName)
		.or(process.env['HOME'] + '/.local/share', appName).result;
	break;
```

And change them to:

```
case 'linux':
	dir = prepareDir("/tmp", "arduino_ide_2").result;
	break;
```

fstasi, 2022-03-09:
@LordRafa I saw your PR, thanks for that!
@iwmq I confirm that it fixes the problem on linux. The same workaround should be applied on darwin

Sadly, seems like `node-log-rotate` is not maintained anymore. We are currently looking for a better replacements. If that is not feasable we will fork it and apply the patches on our end

kittaakos, 2022-08-04:
> The same workaround should be applied on darwin

@fstasi, do we have a concrete bug report on macOS?

PR #1288 (2022-08-04): Set `XDG_CONFIG_HOME` env on Linux when not set.

### Motivation

Set the `XDG_CONFIG_HOME` env on Linux when not set. Otherwise, `node-log-rotate` creates a folder with the `undefined` name.

### Change description

Patched the patched `main.js` for the Theia backend. 🤦 

### Other information

Closes #394.

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated  [... cut]
commit 0b33b517 (2022-08-04): Set `XDG_CONFIG_HOME` env on Linux when not set.

Otherwise, `node-log-rotate` creates a folder with `undefined` name.

Closes #394.
closed 2022-08-04 by kittaakos